prusa3d/PrusaSlicer #7235: MMU Painting still creating dimples in exterior perimeter

jb221, 2021-11-03:
### Version
2.4.0-beta1+win64

### Operating system type + version
Windows 10 64-bit

### Behavior
* Same as #7104, the dimples are greatly reduced on very regular geometries (spheres and cylinders) but are still present on other geometries
* These dimples are not created when slicing using a single color **that matches the "default" extruder.** If the model is a single color but the default extruder color is set to a different extruder you will still get these dimples.
* I compiled a build based on the code last night to test the fix for #7112 (which is indeed fixed) and that's when I noticed the dimples are still present.
* I've had this happen on almost every model I've checked, whether I've ran them through Netfabb or not.

#### Project File (.3MF) where problem occurs
[Dimple Example.zip](https://github.com/prusa3d/PrusaSlicer/files/7470916/Dimple.Example.zip)

With MMU Painting
[image]

Sliced in Single Mode
[image]

hejllukas, 2021-11-05:
Thank you for reporting this issue. It should be fixed with f5a6e532982b42c36f3a39b02353c5a44325326f. I tested it on various models, and for them, dimples are gone. Could you please verify that this [build](https://drive.google.com/file/d/1YaVtgg-ySB5X2m8tdwZODIXFJPVQG7Vr/view?usp=sharing) (or you can compile the current master branch) fixes dimples also on your other models?

jb221, 2021-11-06:
That seems to have fixed it! Thank you so much!

closed 2021-11-06 by jb221