prusa3d/PrusaSlicer #7238: Height of the first layer of the print object when using rafts.

viper0078, 2021-11-04:
### Version
2.4 beta1

### Operating system type + version
windows10 64bit

### 3D printer brand / version + firmware version (if known)
snapmaker2.0

### Behavior
When using rafts, I don't think the effect of rafts contact Z distance on the modeling is taken into account.
The slice result is as follows.
[image]

In the actual print, the first layer of the object will be higher by the amount of raft contact Z distance.
Therefore, the object will also be higher.
[image]

I would like you to consider addressing this issue.

rtyr, 2021-11-04:
I am afraid I don't understand your issue. Please provide some example 3MF project file.

viper0078, 2021-11-04:
I printed a 10x10x10mm cube.
Layer Height = 0.32mm
Raft Contact Z dictance=0.24mm

[image]

The horizontal size error is very small.
[image]

Vertical size error occurs at 0.4 mm.
[image]

I tried printing only the first layer of the object with the same print settings.
[image]

The thickness of the first layer appears to be about 0.55mm.
[image]

I think it is because the height of the first layer of the printout is
 (LayerHeight+RaftContactZdistance).

I think this is one of the reasons for the error in the height of the printed material.

schaudi, 2021-11-04:
Of course the liquid plastic, flowing out of the nozzle is filling the hole 0.56mm under the nozzle. That's just physics u can't overturn. The amount of liquid prusaslicer is calculating is correct - if it were the way u descriped in the first post, the lines would be closed. that's why u regularly make that distance as small as possible. if u want perfect vertical accuracy u have to use ZERO or no raft at all.

viper0078, 2021-11-05:
I use Raft for all my printing for several reasons.
RaftContactZdistance sets the distance at which the print is firmly attached to the raft and can be easily removed.

Here is a suggestion on how to correct the raft contact Z distance.
Let's assume the slice parameters are as follows
Layer Height = 0.24mm
raft contact Z distance = 0.1mm

In the current version, is the height to slice the bottom of the print 0.12mm?　(0.24/2 = 0.12mm)
If so, then the bottom will be sliced at a height of 0.12+0.1=0.22mm to compensate for RaftContactZdistance.
Since the LayerHeight is still 0.24mm, 0.1mm of the bottom surface will not be sliced.

There may be a better way to compensate for this.

If you don't use a raft, the height error is the gap between the print bed and the print nozzle. In my case, I use rafts, so this case does not bother me.

n8bot, 2021-11-05:
It seems like you do not want a raft z separation on first layer. Instead, you want no z separation, but slightly underextruded first object layer over the raft. That way, the object can be more-easily removed from the raft, but the print height of the object layers are not shifted. This way, the bottom (first) object layer will have the same look and behaviour as with Z separation, but the Z height of the object will be more accurate.

viper0078, 2021-11-05:
That's an interesting idea.
I'd like to try it, but the current version doesn't allow you to change the flow rate of the first layer, does it?
However, if the LayerHeight is small, I am worried that lowering the flow rate will reduce the strength of the first layer.
I won't know which idea is better until I try it.

rtyr, 2021-11-05:
Related issue https://github.com/prusa3d/PrusaSlicer/issues/3458.

theely, 2021-11-07:
@rtyr  not sure this is related with #3458.

This issue is not related to bridging it's a raft implementation choice that causes the first layer to expand beyond the object desired geometry.

Instead of lowering the Raft by the contact Z distance it should be the first layer that is raised by the Z distance. This way the geometry is preserved while reducing the pressure between the layers.

Cura for instance does exactly that, which results in precise printing and easy to remove Raft.
[image]

viper0078, 2021-11-08:
I don't think it is related to #3458 either.
theely, have you tried to use 3.4beta1?
RaftContactZdistance in 3.4beta1 is similar to RaftAirGap in Cura.
This distance can cause errors in the height of the print. I hope the development team will consider a solution to this problem.

closed 2026-09-29 by prusaslicer-bot[bot]