prusa3d/PrusaSlicer #7239: bridge flow ratio

viper0078, 2021-11-04:
### Version
2.4beta1

### Operating system type + version
win10 64bit

### 3D printer brand / version + firmware version (if known)
snapmaker2.0

### Behavior
I don't think the bridge flow ratio is functioning correctly.

Layer height 0.24mm
Top contact Z distance 0.12mm
bridge flow ratio 1.0

[image]

[image]
The height of the bridge layer is large, as seen in the red circle.

neophyl, 2021-11-04:
Thats the normal operation for bridges in previous versions.  Its not bridge flow.  As the extrusion isnt being squished against a lower layer the height is assumed to be your nozzle width, hence the thick layer.  Its always been that way.

However you seem to be using the 2.4 release (which the release notes specifically mention this function change), that has the new option under print settings>layers and perimeters>Quality>Thick Bridges.  By default it ticked to keep the current behaviour but if you untick it the logic is altered.

bubnikv, 2021-11-04:
explained by @neophyl , closing

closed 2021-11-04 by bubnikv