prusa3d/PrusaSlicer #7240: top contact Z distance

viper0078, 2021-11-04:
### Version
2.4beta1

### Operating system type + version
windows10 64bit

### 3D printer brand / version + firmware version (if known)
snapmaker2.0

### Behavior
Top contact Z distance does not seem to be working.
[image]
Layer height = 0.24mm
top contact Z distance = 0.14mm
top interface layers = 2
[image]

I think the 0.16mm layer is the adjustment layer for the Top contact Z distance. However, it does not work because there is no support in the lower part of the bridge.
[image]

The height of the support interface layer is 9.68mm.
The height of the bridge layer is 10.16mm.
The distance between the support interface layer and the bridge layer is
10.16-9.68-2.4(LayerHeight)=0.3mm

If an adjustment layer of 0.16mm is created under the bridge, then

0.3mm - 0.16mm = 0.14mm
This will match the value set for the top contact Z distance.

I think it is a bug.

rtyr, 2021-11-04:
That is a correct/expected behavior when the "thick bridges" option is enabled. In such case, bridge is printed as a round extrusion with the diameter equal to the nozzle diameter (independent from layer height used on the object). See the screenshot here https://github.com/prusa3d/PrusaSlicer/issues/4953#issuecomment-716057549.

If you disable "thick bridges" option, bridges will use regular flow, similar to other layers. That is probably the behavior you are looking for.

Closing.

closed 2021-11-04 by rtyr

viper0078, 2021-11-04:
Thanks for the reply.
I got the ideal behavior with the thick bridges option.