prusa3d/PrusaSlicer #7241: Prusaslicer problem sending file to Octoprint

Vosmash, 2021-11-04:
### Version
e 2.3.3

 Verze 2.3.3+win64

Windows 10 (build 19043), 64-bit edition

_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
Ender 3 PRO

### Behavior
* _Describe the problem_Not able to upload file to Octoprint. Printer is set properly, Test communication with Octoprint works, Octoprint via web is accessible
* _Steps needed to reproduce the problem_Just to try to send it
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_
[image]

_Is this a new feature request? [no](url)

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_
PrusaSlicer
Version:   2.3.3+win64
Build:     PrusaSlicer-2.3.3+win64-202107161027

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 19043), 64-bit edition
Total RAM size [MB]: 17,118MB
OpenGL installation
GL version:   4.6.0 NVIDIA 471.96
Vendor:       NVIDIA Corporation
Renderer:     NVIDIA GeForce GTX 1060 6GB/PCIe/SSE2
GLSL version: 4.60 NVIDIA
Installed extensions:

GL_AMD_multi_draw_indirect
GL_AMD_seamless_cubemap_per_texture
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_compute_variable_group_size
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
[... 337 lines cut ...]
GL_NV_timeline_semaphore
GL_NV_transform_feedback
GL_NV_transform_feedback2
GL_NV_uniform_buffer_unified_memory
GL_NV_vertex_array_range
GL_NV_vertex_array_range2
GL_NV_vertex_attrib_integer_64bit
GL_NV_vertex_buffer_unified_memory
GL_NV_vertex_program
GL_NV_vertex_program1_1
GL_NV_vertex_program2
GL_NV_vertex_program2_option
GL_NV_vertex_program3
GL_NV_viewport_array2
GL_NV_viewport_swizzle
GL_OVR_multiview
GL_OVR_multiview2
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_lod
GL_SGIX_depth_texture
GL_SGIX_shadow
GL_SUN_slice_accum
GL_WIN_swap_hint
WGL_EXT_swap_control

kocikdav, 2021-11-04:
Thank you for reporting. How did you setup your address to the octoprint? IP address or hostname? If hostname does IP fails too?

kocikdav, 2021-11-04:
Also I see you are using version 2.3.3. Please try our latest beta first. There has been changes in file uploading.

Vosmash, 2021-11-04:
Sounds good, thanks! Where exactly to download?

Vosmash, 2021-11-04:
Well, the problem is still the same ... there must be another issue.

PrusaSlicer
Version:   2.4.0-beta1+win64
Build:     PrusaSlicer-2.4.0-beta1+win64-202110251333

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 19043), 64-bit edition
Total RAM size [MB]: 17,118MB
OpenGL installation
GL version:   4.6.0 NVIDIA 471.96
Vendor:       NVIDIA Corporation
Renderer:     NVIDIA GeForce GTX 1060 6GB/PCIe/SSE2
GLSL version: 4.60 NVIDIA
[image]

Vosmash, 2021-11-04:
There is a IP adress in a printer setup. Test work in a both versions ....

kocikdav, 2021-11-04:
Ok thanks, is there anything special about your Octoprint? For example some people had troubles when using self-signed SSL certificate. but that should be fixed..

Vosmash, 2021-11-04:
No, Octoprint is standard one, on Raspi. It worked in the morning and after that it started to make problems - no changes at all, after one print I couldn´t do next one. I am not sure, where the problem is right now.

bubnikv, 2021-11-04:
Did it ever work for you with any PrusaSlicer version?
Does it work for you with any other application? For example Cura with some plugin?
IMHO the error says it all: Something is broken on your side.

Vosmash, 2021-11-07:
The problem was in Octoprint, after reinstalation it works.

closed 2021-11-07 by Vosmash

Matoni77, 2021-11-08:
Dobrý deň.Prepáčte ale som v koncoch.Pokiaľ som si to vedel odstránit tak som nikoho s otázkami neobťažoval.Používam PrusaSlicer verziu 2.3.0+win64.Po slicovani objektu chcem poslat gcode do tlačiarne.Následne mi vyskočí chyba HTTP 500.Posledné čo som v programe spravil bola aktualizácia nastavaní,ktorá ma "šikanovala" pri spustení programu.Za skorú odpoveď ďakujem.

bubnikv, 2021-11-08:
Bohužel vám neporadíme. Zkuste přeinstalovat OctoPrint? Zkuste se zeptat člověka v išů #7253.