prusa3d/PrusaSlicer #7242: thick bridges - regression on tight tolerances

theely, 2021-11-04:
### Version
2.4.0 - beta1

### Operating system type + version
MacOS

### 3D printer brand / version + firmware version (if known)
Ender 3 Pro - Marlin 2

### Behavior
When thick bridges is enabled small tolerances like M3 screw holes are no longer respected. The additional thickness is adding too much material.

With thick bridges enabled
[image]

Without thick bridges enabled
[image]

While I see the benefit of having more robust bridges it should not come at the cost of non respecting the model geometry.
I suggest to tun off this setting by default. 

@bubnikv FYI

neophyl, 2021-11-04:
the reason its on by default is that the old old behaviour before the option of thin bridges was added in the 2.4 release.  Its usually not done to change existing expected behaviour.

rtyr, 2021-11-04:
> When thick bridges is enabled

When enabled, PrusaSlicer prints bridges in the same old way, which was always used. That is also the reason, why it's enabled by default (if not set differently in the configuration bundle), as neophyl mentioned. 

Also, disabling this parameter usually requires other additional changes (contact Z distance, bridge flow ratio, bridging speed). For example, we have this option disabled by default for our printers now, but it required lot of other changes related to it. 

Answered. Closing.

closed 2021-11-04 by rtyr

rtyr, 2021-11-05:
Related issue https://github.com/prusa3d/PrusaSlicer/issues/3458.