prusa3d/PrusaSlicer #7243: Cut causes "open edges"

JohnOCFII, 2021-11-04:
### Version
Version 2.4.0-beta1+arm64
### Operating system type + version

MacOS 12.0.1 Monterey  on new MacBook Pro M1 Pro

### 3D printer brand / version + firmware version (if known)

Prusa MK3S+ MMU2S and firmware 3.10.0

### Behavior

PrusaSlicer is displaying "Remaining 138 open edges" as well as the triangle symbol for a broken model after using the `Cut` function to cut the model in half, an retaining just the lower half of the model.

This is a simple model I use often for testing and tuning.  I did not see this issue in the 2.4.0 alpha releases, but I switched from Intel MacBook Pro to M1 MacBook Pro since last week.

[image]

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_
[SquaredCircle 50mm-badcut.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/7478236/SquaredCircle.50mm-badcut.3mf.zip)

bubnikv, 2021-11-05:
@FidelCapo would you please test on M1?

This code was refactored recently and we had such issue, we hoped we fixed them. Actually 2.3.3 created such cracks as well, but it then ran the triangle fixing routine to clean them up. In 2.4.0-beta1 we hoped we generate the mesh correctly up front, thus we don't run the triangle mesh fixing anymore. Also in 2.4.0-beta1 the mesh diagnostics was improved as well.

FidelCapo, 2021-11-05:
@JohnOCFII Can you please attach original STL file? Thank you

JohnOCFII, 2021-11-05:
@FidelCapo  -- here is the zipped STL.
[SquaredCircle 50mm.stl.zip](https://github.com/prusa3d/PrusaSlicer/files/7486724/SquaredCircle.50mm.stl.zip)

slomomojo, 2022-06-27:
This still exists in PrusaSlicer 2.4.2 (Arch Linux).  All you need to do is Add Shape > Sphere then Cut to get the message.  Does not seem to impact printing but it makes you think there is something wrong with the model.

arcivanov, 2023-01-05:
Still exists in PS 2.5 on Linux.

sp847242, 2023-10-12:
I've also gotten this on multiple models on 2.6.1. I don't remember when I started seeing it, but it was before 2.6.1.
The models import fine, no open edges reported, no errors at all.
After the cuts, models are sometimes fine, or sometimes can have several dozen open edges.

In many cases, it still manages to slice and print fine. But in others, the sliced model is unprintable, as its "first" layer might be quite a ways above the bed because of various missing triangles. 
[image]
[image]

I also just tried out JohnOCFII's SquaredCircle file, cut it in the middle, and got 3 open edges on one piece, and 6 open edges on the other.
But it seems that if I cut anywhere but center, even just 25.01mm, then I do not get any open edges.

FidelCapo, 2024-01-17:
This issue should be fixed and the fix will be part of the next release.
Closing.

closed 2024-01-17 by FidelCapo

j28CLY, 2025-12-04:
> This issue should be fixed and the fix will be part of the next release. Closing.

I am using v2.9.4-rc1 and I have continued to get this. 
If a model is sliced exactly in half, it is prone to showing the message "open edges or errors were detected after the cut."
For example, a part that's 5in/127mm tall: If the cutting plane is at 2.50in/63.50mm, both resulting parts have open edges. If cut at 2.51in, or if done at 63.51mm, both parts are fine. Even just that 0.01mm is enough to prevent this.

I just quick made a part which shows the error when cut.
I mostly work with thin-walled parts with small rounds, and a high triangle count. One of those things might contribute to this error showing up.
- If I cut the part at 63.50mm, one part has 42 open edges, and the other has 21.
- If I cut it at 63.51mm, both parts are fine.
[Part1.zip](https://github.com/user-attachments/files/23942748/Part1.zip)