prusa3d/PrusaSlicer #7245: 2.4.0-beta1 force close when resize object to 0

SamLefebvre, 2021-11-05:
### Version
2.4.0-beta1+win64

### Operating system type + version
Windows 10 (build 19042), 64-bit edition

### Behavior
* _Describe the problem_
PrusaSlicer force close without warning.

* _Steps needed to reproduce the problem_

1. Add shape, Box.
2. In object manipulation, set the size of X to 0.
3. Click somewhere else to exit the textbox.

* _Expected Results_

Not closing, like version 2.3.3.

* _Actual Results_
The application is closing, without warning or saving prompt.

A screenshot of where the bug occur:
[image]

_Is this a new feature request?_
No.

combolek, 2021-11-06:
@enricoturri1966 I think this is a regression in 4521945bb3aaafdc63b4ed54c03c753dfa08594c
```
commit 4521945bb3aaafdc63b4ed54c03c753dfa08594c
Author: enricoturri1966 <enricoturri@seznam.cz>
Date:   Mon Oct 4 14:07:45 2021 +0200

    Tech ENABLE_OUT_OF_BED_DETECTION_IMPROVEMENTS - Faster printbed collision detection using the new function Geometry::intersect()
```

lukasmatena, 2021-11-12:
This should be fixed in PrusaSlicer 2.4.0-beta2 (will be released shortly). Closing.

closed 2021-11-12 by lukasmatena