prusa3d/PrusaSlicer #7246: Error when resize to bed limit, Failed processing of the output_filename_format template

SamLefebvre, 2021-11-05:
### Version
2.4.0-beta1+win64

### Operating system type + version
Windows 10 (build 19042), 64-bit edition

### Behavior
* _Describe the problem_
When I click to Export G-code, PrusaSlicer has encountered an error.

* _Steps needed to reproduce the problem_

Steps for Original Prusa i3 MK3S & MK3S+

1. Add shape, Box
2. Click on Slice now
3. Enter 210 as the size for X
4. Click on Export G-code

Steps for Creality Ender-3 V2
1. Add shape, Box
2. Click on Slice now
3. Unlock the size ratio
3. Enter 250 as the size for Z
4. Click on Export G-code

* _Expected Results_
No error.

* _Actual Results_

```
Failed processing of the output_filename_format template.
Parsing error at line 1: Not a variable name
{input_filename_base}_{layer_height}mm_{filament_type[0]}_{printer_model}_{print_time}.gcode
 ^
```

[image]

_Is this a new feature request?_
No.

bubnikv, 2021-11-08:
>    4. Enter 250 as the size for Z
>    5. Click on Export G-code

Don't know what's wrong, works for me. Please attach a 3MF.

SamLefebvre, 2021-11-08:
Here is a 3MF with a box. When I open the file, the bug does not occur. 
[Shape-Box.zip](https://github.com/prusa3d/PrusaSlicer/files/7498750/Shape-Box.zip)

I also attached a gif showing step by step how I can reproduce the bug with a new project.

[image]

lukasmatena, 2021-11-09:
@SamLefebvre I managed to reproduce the issue in 2.4.0-beta1, but not in current master. It was probably fixed during recent improvements in PlaceholderParser. Closing. Please, retest in beta2 (after it is released) and let us know if the issue persisted. Thanks.

closed 2021-11-09 by lukasmatena