prusa3d/PrusaSlicer #7247: Wxwidgets 3.0 yet again.  MsgDialog.hpp:270:40: error: ‘wxASCII_STR’ was not declared in this scope

clytle374, 2021-11-05:
Gentoo linux, building from current git source.  WxWidgets 3.2 still hasn't been released, I'm starting to think it's a unicorn.
Thanks if you can fix it, if not I guess it's time to remove the wx stable from the build.
  `[ 53%] Built target imgui
Consolidate compiler generated dependencies of target libslic3r_gui
[ 53%] Building CXX object src/slic3r/CMakeFiles/libslic3r_gui.dir/GUI/Gizmos/GLGizmoSlaSupports.cpp.o
In file included from /home/cory/sources/prusa/src/slic3r/GUI/Gizmos/GLGizmoSlaSupports.cpp:21:
/home/cory/sources/prusa/src/slic3r/GUI/MsgDialog.hpp:270:40: error: ‘wxASCII_STR’ was not declared in this scope
  270 |                 const wxString& name = wxASCII_STR(wxStaticLineNameStr))
      |                                        ^~~~~~~~~~~
make[2]: *** [src/slic3r/CMakeFiles/libslic3r_gui.dir/build.make:397: src/slic3r/CMakeFiles/libslic3r_gui.dir/GUI/Gizmos/GLGizmoSlaSupports.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:1222: src/slic3r/CMakeFiles/libslic3r_gui.dir/all] Error 2
make: *** [Makefile:146: all] Error 2
`
Thanks again
Cory

combolek, 2021-11-06:
Can you try this?

```diff
diff --git a/src/slic3r/GUI/MsgDialog.hpp b/src/slic3r/GUI/MsgDialog.hpp
index 909ff396d..a4807fd06 100644
--- a/src/slic3r/GUI/MsgDialog.hpp
+++ b/src/slic3r/GUI/MsgDialog.hpp
@@ -98,7 +98,7 @@ public:
                                const wxPoint& pos = wxDefaultPosition,
                                const wxSize& size = wxDefaultSize,
                                long style = wxLI_HORIZONTAL,
-                               const wxString& name = wxASCII_STR(wxTextCtrlNameStr))
+                               const wxString& name = wxString::FromAscii(wxTextCtrlNameStr))
        : wxTextCtrl(parent, id, wxEmptyString, pos, size!=wxDefaultSize ? size : (style == wxLI_HORIZONTAL ? wxSize(10, 1) : wxSize(1, 10)), wxSIMPLE_BORDER, wxDefaultValidator, name)
        {
                this->Enable(false);
@@ -268,7 +268,7 @@ public:
                const wxPoint& pos = wxDefaultPosition,
                const wxSize& size = wxDefaultSize,
                long style = wxLI_HORIZONTAL,
-               const wxString& name = wxASCII_STR(wxStaticLineNameStr))
+               const wxString& name = wxString::FromAscii(wxStaticLineNameStr))
                : wxStaticLine(parent, id, pos, size, style, name) {}
        ~StaticLine() {}
 };
```

slmb3, 2021-11-06:
@combolek yes it works

combolek, 2021-11-06:
OK, thanks for testing. I submitted PR #7256 for this.

clytle374, 2021-11-08:
So

closed 2021-11-08 by clytle374

lukasmatena, 2021-11-08:
I merged the PR, this issue should be fixed.