prusa3d/PrusaSlicer #7248: Segfault when using 2.4 beta1 one on macOS Monterey

nsheridan, 2021-11-05:
### Version
2.4.0 beta1

### Operating system type + version
macos monterey 12.1 beta (21C5021h) running on a Macbook Air M1

### 3D printer brand / version + firmware version (if known)
Prusa mk3s+ 3.10.0

### Behavior
Right after I send a sliced model to octoprint, PS will segfault. I've attached the crash report:

```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               PrusaSlicer [31720]
Path:                  /Applications/PrusaSlicer.app/Contents/MacOS/PrusaSlicer
Identifier:            com.prusa3d.slic3r/
Version:               PrusaSlicer PrusaSlicer-2.4.0-beta1+x64-202110251325 (PrusaSlicer-2.4.0-beta1+x64-202110251325)
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2021-11-05 19:31:05.8575 +0000
OS Version:            macOS 12.1 (21C5021h)
Report Version:        12
Anonymous UUID:        FF465A61-8784-DEF1-651D-352A570ED612

Sleep/Wake UUID:       F1F5A62D-CAB7-4CF4-8DF9-252B847107EF

Time Awake Since Boot: 100000 seconds
Time Since Wake:       34614 seconds

System Integrity Protection: enabled

Crashed Thread:        0  slic3r_main  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000098
Exception Codes:       0x0000000000000001, 0x0000000000000098
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [31720]

VM Region Info: 0x98 is not in any region.  Bytes before following region: 4366827368
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      __TEXT                      104488000-105bb4000    [ 23.2M] r-x/r-x SM=COW  ...S/PrusaSlicer

Kernel Triage:
VM - pmap_enter failed with resource shortage
VM - pmap_enter failed with resource shortage
VM - pmap_enter failed with resource shortage
VM - pmap_enter failed with resource shortage
VM - pmap_enter failed with resource shortage

Thread 0 Crashed:: slic3r_main Dispatch queue: com.apple.main-thread
0   PrusaSlicer                   	       0x104df55f0 Slic3r::GUI::NotificationManager::update_notifications(Slic3r::GUI::GLCanvas3D&) + 604
1   PrusaSlicer                   	       0x104df55f0 Slic3r::GUI::NotificationManager::update_notifications(Slic3r::GUI::GLCanvas3D&) + 604
2   PrusaSlicer                   	       0x104ab485c Slic3r::GUI::GLCanvas3D::on_idle(wxIdleEvent&) + 220
3   PrusaSlicer                   	       0x105788388 wxEvtHandler::SearchDynamicEventTable(wxEvent&) + 228
4   PrusaSlicer                   	       0x105788084 wxEvtHandler::ProcessEventLocally(wxEvent&) + 88
5   PrusaSlicer                   	       0x105787ee4 wxEvtHandler::ProcessEvent(wxEvent&) + 56
6   PrusaSlicer                   	       0x1057884e8 wxEvtHandler::SafelyProcessEvent(wxEvent&) + 24
7   PrusaSlicer                   	       0x1055f5468 wxWindowBase::SendIdleEvents(wxIdleEvent&) + 92
8   PrusaSlicer                   	       0x1055f5490 wxWindowBase::SendIdleEvents(wxIdleEvent&) + 132
[... 419 lines cut ...]
Display: DELL U2719DC, 2560 x 1440 (QHD/WQHD - Wide Quad High Definition), Main, MirrorOff, Online
Memory Module: LPDDR4
AirPort: Wi-Fi, wl0: Sep 13 2021 21:06:59 version 18.20.310.14.7.8.119 FWID 01-9d72ee4a
Bluetooth: Version (null), 0 services, 0 devices, 0 incoming serial ports
Network Service: USB 10/100/1000 LAN, Ethernet, en7
Network Service: Wi-Fi, AirPort, en0
USB Device: USB31Bus
USB Device: USB3.0 Hub
USB Device: USB 10/100/1000 LAN
USB Device: 4-Port USB 3.1 Hub
USB Device: USB3.0 Card Reader
USB Device: USB2.0 Hub
USB Device: 4-Port USB 2.1 Hub
USB Device: Keychron K8
USB Device: Anker USB-C Hub Device
USB Device: USB31Bus
USB Device: hub_device
USB Device: hub_device
USB Device: USB Receiver
USB Device: HD Pro Webcam C920
USB Device: YubiKey OTP+FIDO+CCID
USB Device: composite_device
Thunderbolt Bus: MacBook Air, Apple Inc.
Thunderbolt Bus: MacBook Air, Apple Inc.
```

bubnikv, 2021-11-08:
This looks like the notification crash that we have in beta1 and which is
already fixed on master.

plucker, 2021-11-17:
I'm not sure if this was the same as I was seeing but I was getting a lot of crashes in this beta version, 2 or 3 a day.

lukasmatena, 2021-11-18:
@nsheridan Thanks. This is probably indeed a duplicate of #7170. If you still see it in beta2, let us know. Closing the issue for now (and hopefully forever).

@plucker If there is another crash that you're able to reproduce, we will be thankful if you could tell us.

closed 2021-11-18 by lukasmatena