prusa3d/PrusaSlicer #7249: Estimated time and filament usage not visible for GCode from PrusaSlicer 2.4 beta1

NobodysNightmare, 2021-11-06:
### Version
```
Version 2.4.0-beta1+win64
```

### Operating system type + version
* Windows 10

### 3D printer brand / version + firmware version (if known)
* Prusa Mini+
* FW: 4.3.3

### Behavior
The printer does not indicate estimated time and estimated filament usage during print preview for GCode created by PrusaSlicer 2.4. GCode created through 2.3 works fine on the same printer, so I suppose the issue is with the generated GCode file:

[image]

#### Project File (.3MF) where problem occurs

[Tower_keyed_150mm_0.2mm_PLA_MINI_13h11m.gcode.zip](https://github.com/prusa3d/PrusaSlicer/files/7489953/Tower_keyed_150mm_0.2mm_PLA_MINI_13h11m.gcode.zip)

I do not have a 3MF of the given project ready, but the issue does not seem to be specific to that one project. I set-up the Prusa Mini as the target printer during setup.

neophyl, 2021-11-06:
https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/1646

rtyr, 2021-11-06:
> prusa3d/Prusa-Firmware-Buddy#1646

Closing.

closed 2021-11-06 by rtyr