prusa3d/PrusaSlicer #7250: Nozzle sent 7 layers into printed object, after print done.

fungreenfox, 2021-11-06:
### Version
_Version of PrusaSlicer 2.3.3+Win64_

### Operating system type + version
_Windows 10 64bit
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_
PrusaSlicer
Version:   2.3.3+win64
Build:     PrusaSlicer-2.3.3+win64-202107161027

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 19043), 64-bit edition
Total RAM size [MB]: 51,449MB
OpenGL installation
GL version:   4.6.0 NVIDIA 496.13
Vendor:       NVIDIA Corporation
Renderer:     NVIDIA GeForce RTX 2060/PCIe/SSE2
GLSL version: 4.60 NVIDIA
Installed extensions:

GL_AMD_multi_draw_indirect
GL_AMD_seamless_cubemap_per_texture
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_compute_variable_group_size
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_fragment_shader_interlock
GL_ARB_framebuffer_no_attachments
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
[... 348 lines cut ...]
GL_SUN_slice_accum
GL_WIN_swap_hint
WGL_EXT_swap_control

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on. Creality 6 with Octoprint.

### Behavior
When printing is done, the nozzle is sent about 7-10 layers down into the print, and burning its way.
Tried to add a step 20 down to hotbed after printjob is done, into Octoprint g-code boxes:
**
G91;Set to Relative Positioning
G1 Z40
G90;Set to Absolute Positioning,
 ** 
but it seems Prusa
slicer getting there first, getting its code executed before anything can be done.

_Is this a new feature request?_ This is more like an improvement of allowing g-code to override this fail.

#### Project File (.3MF) where problem occurs
Was not able to upload anything. Even picture fails uploading here.

_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

rtyr, 2021-11-06:
Please provide your 3MF project file (you have to zip it first).

> Upload a PrusaSlicer Project File (.3MF) (Plater -> Export plate as 3MF for Slic3r PE 1.41.2 and older, File -> Save / Save Project for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)
> Images (PNG, [...]

fungreenfox, 2021-11-07:
The upload file manager at this page, does only accept video file, so how to upload a .3MF file?

neophyl, 2021-11-07:
Video is of no use in figuring out what is causing the issue, take your saved 3mf file, zip it up and then it will attach here.

rtyr, 2021-11-07:
The 3MF you provided is Cura project file, please provide your PrusaSlicer project file (File - Save project as...).

Your issue is likely not PrusaSlicer related, but we need some data to check.

fungreenfox, 2021-11-07:
> The 3MF you provided is Cura project file, please provide your PrusaSlicer project file (File - Save project as...).
> 
> Your issue is likely not PrusaSlicer related, but we need some data to check.

I have updated the file and linked again in the comment above.

fungreenfox, 2021-11-07:
This issue is not observed by using creality slicer or ultimaker, and i have never seen it before, only by prusa slicer.

neophyl, 2021-11-07:
I saw that in your filament settings too which is strange as when you combine it with the custom gcode in the printer settings it doesn't make sense

G92 E0
G1 Z28.050 F6000.000
M107  **<---end of normal printing for the cylinder object**
;TYPE:Custom
; Filament-specific end gcode 
;END gcode for filament  **<-- this is your section as defined under your filament end gcode**
G91;Set to Relative Positioning
G1 Z40
G90;Set to Absolute Positioning

So in effect it moves it up another 40mm from whatever height it ended up at.  Thats ok I follow that.

This section is added to the gcode as its defined under Printer Settings>custom gcode> end gcode
M104 S0 ; turn off temperature
;G28 X0  ; home X axis
G1 Z5 F5000 ; lift nozzle  **<-- as the machine has been put into Absolute positioning mode due to your filament end code wont this just cause the extruder to drop down to a height of 5mm ?  It would raise up another 5mm except that you put it into absolute mode with the G91.  As you haven't moved the extruder in x/y I would expect this to drop the still hot extruder into your printed part with nasty consequences as it tries to get down to 5mm height **
M84     ; disable motors

fungreenfox, 2021-11-07:
It is possible you are right about this. In Octoprint, to make that work, it needs g91 relative pos, in order to send the bed downwards, and then G90 absolute pos. in order to set the positioning back to what it was. So the Otctoprint might work with this in another way than prusa does?
Glad you pointed this out, and that section with g-code in printer settings has been unseen my me, as i only saw that kinda section in filament settings. Ill try remove the G1 Z5 F5000 ; lift nozzle and find out.

neophyl, 2021-11-07:
What printer do you have ?  Normally  a G1 Z20 will move the extruder UP (or the bed down if your printer controls Z by moving the bed)  As that is INCREASING the distance between the current position and where you have told it to go.  The only time Ive seen otherwise is when a stepper was wired in reverse.  Not that Ive seen all printer types though.

Sounds like you need to check all the basic movement assumptions first for configuring your printer.

fungreenfox, 2021-11-07:
Its a Creality Ender 6.

fungreenfox, 2021-11-07:
Moving g90 behind the other code, makes the nozzle (at finish printing), move out to the side, and first then move the bed down. The home x axis code is not removed, but extruder doesnt go the home position, but just to the side. Anyways, i'm happy, as it doesn't move into the object.

rtyr, 2021-11-08:
Seems resolved, can we close this issue?

fungreenfox, 2021-11-08:
Yes, thank you for helping out.

closed 2021-11-08 by fungreenfox