prusa3d/PrusaSlicer #7251: extrudeur rien que pour support

18pool49, 2021-11-06:
[image]
### Version
_Version of PrusaSlicer used goes here_     2.3.3

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version   w10 
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)   tenlog TL-D3 PRO   Marlin 
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_         
* c'est une remarque : 
* sur le choix qui est PAS donné d'un extrudeur RIEN que pour les "SUPPORTS" 
* qui est actuellement confondu avec >bordure , jupe et radeau qui a rien a voir avec des supports surtout en ce qui concerne la qualité matière 
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

rtyr, 2021-11-08:
I don´t understand your issue or what are you trying to achieve. Please try to describe it in English, preferably with some example 3MF project file.

18pool49, 2021-11-09:
bonjour 
le support ou interface du support est réalisé avec une matière bien spécifique (PVA / Hips) ou même un PETG en interface du support avec PLA en matière pour le support  et pour la pièce  ; mais avec Prusa ! ! !  "l'interface support"  qui est bien présente dans paramètre des supports est réalisé avec le même extrudeur que le radeau (radeau qui est conçu dans la même matière que la pièce pour assurer une bonne fixation sur le lit ; 
donc le support ou interface devrait avoir le choix d'un extrudeur avec la matière qui lui est spécifique a lui seul et ne pas être confondu avec > la jupe ,la bordure ou le radeau  comme c'est le cas actuellement dans Prusa 
[image]

bubnikv, 2021-12-08:
> I don´t understand your issue or what are you trying to achieve. Please try to describe it in English, preferably with some example 3MF project file.

No data provided, closing.

closed 2021-12-08 by bubnikv