prusa3d/PrusaSlicer #7252: différence de température sur "IDEX" impossible

18pool49, 2021-11-06:
### Version
_Version of PrusaSlicer used goes here_  2.3.3

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version w10 
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known) tenlog TLD3 PRO 
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_  il est impossible de programmer deux températures différente entre les deux extrudeurs R/L   
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

rtyr, 2021-11-08:
```
M104 S{first_layer_temperature[0]} T0 ; left
M104 S{first_layer_temperature[1]} T1 ; right
```

closed 2021-11-08 by rtyr

18pool49, 2021-11-09:
merci vous êtes bien aimable ,je sais très bien que Marlin dispose du "M104 S xxxx "  
mais le trancheur n'en propose que un (1) "M104 S xxxx" pour le T0 et rien d'autre 
pour le T1 il y a pas de possibilité il est invisible a l'écran bien que dans les paramètre machine il y a bien d'inscrit  (deux = 2 ) extrudeurs 
alors ma question est comment je fait pour rendre visible cette fonction "M104 Sxxxx" pour T1  
slts