prusa3d/PrusaSlicer #7253: Chyba při nahrávání do tiskového serveru: HTTP 500: {"error":"The server encountered an internal error and was unable to complete your request. Either the server is overloaded or there is an error in the application."}

Matoni77, 2021-11-06:
### Version
_Version of PrusaSlicer used goes here_

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

bubnikv, 2021-11-08:
Co to je za verzi octoprintu? Máte někde povolenou češtinu? Na octoprintu? Na Windowsech nebo z čeho pouštíte slicer?

bubnikv, 2021-11-08:
Sounds like a duplicate of 
https://github.com/prusa3d/PrusaSlicer/issues/7241#issuecomment-962636727

Matoni77, 2021-11-08:
Dobrý deň.Prepáčte ale som v koncoch.Pokiaľ som si to vedel odstránit tak
som nikoho s otázkami neobťažoval.Používam PrusaSlicer verziu
2.3.0+win64.Po slicovani objektu chcem poslat gcode do tlačiarne.Následne
mi vyskočí chyba HTTP 500.Posledné čo som v programe spravil bola
aktualizácia nastavaní,ktorá ma "šikanovala" pri spustení programu.Za skorú
odpoveď ďakujem.

bubnikv, 2021-12-08:
Bohužel nemůžeme pomoci.

closed 2021-12-08 by bubnikv