prusa3d/PrusaSlicer #7255: Prusaslicer dumps and quits after uploading to octoprint finished

attilah, 2021-11-06:
### Version
2.4.0 beta 1

### Operating system type + version

MacOS Monterey

### 3D printer brand / version + firmware version (if known)

Prusa Mini+ 4.3.3

### Behavior

Octoprint is configured and was working with previous slicer flawlessly, nothing has changed.

After exporting gcode to octoprint prusaslicer unexpectedly quits, while the upload was successful as octoprint even starts to print which completes successfully.

#### Project File (.3MF) where problem occurs

The issue is not tied to a particular file.

mscalora, 2021-11-07:
I've seen this numerous times also, sent the crash dump from my Mac Book Pro / Catalina most times

dartrax, 2021-11-07:
This is probably a duplicate of #7170, which is already fixed.

attilah, 2021-11-07:
I agree @dartrax, closing it as looks like a duplicate even if OS is different.

closed 2021-11-07 by attilah