prusa3d/PrusaSlicer #7260: disabling thick bridges add unnecessary support interfaces

theely, 2021-11-07:
Version
2.4.0 - beta1

Operating system type + version
MacOS

3D printer brand / version + firmware version (if known)
Ender 3 Pro - Marlin 2

Behavior
When thick bridges is disabled unrequited support interfaces are added.

Sliced model with thick bridges enabled:
[image]

Sliced model with thick bridges disabled:
[image]

Project file:
[tb-bug.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/7492486/tb-bug.3mf.zip)

bubnikv, 2021-12-10:
Similar to https://github.com/prusa3d/PrusaSlicer/issues/3538, but there don't seem to be any gap fill. It looks as if the bridging perimeters were not correctly subtracted from overhangs due to some numerical issues.

bubnikv, 2021-12-10:
Improved with 121bb260db5c73c0f65922e1818372006c1f83da

closed 2021-12-10 by bubnikv