prusa3d/PrusaSlicer #7261: Supports not generated for face parallel to build plate in certain part orientations

cuzelac, 2021-11-07:
### Version
Version 2.4.0-beta1+win64

### Operating system type + version
Microsoft Windows 10 Pro
10.0.19043 Build 19043

### 3D printer brand / version + firmware version (if known)
Stock Prusa i3 Mk3S+
3.10.0-4481

### Behavior
* _Description_
    * I have a part with two slots perpendicular to each other. Each slot that has two parallel faces and I need to support the face that is hanging parallel to the build plate. 
    * One slot does not generate supports for enforcers
    * When I orient the part such that the other slot is the one receiving supports, supports are generated properly for enforcers
    * I have included two `.3mf` files that demonstrate both states
    * If 
* _Reproduction steps_
   * I have tried the following and all give the same results:
       * Brush vs. Smart fill enforcers
       * Changing the rotation of the part
       * Changing the support "Pattern angle" setting
   * I can reproduce with a mirror image part that was generated in Fusion 360
   * Further reproduction is beyond my understanding of the slicer.
* _Expected results_
    * Supports for the face parallel to the build plate are generated
    * See file `support-issue-expected-v7.3mf` for the face that works properly 
    * [image]
* _Actual Results_
    * No supports are generated
    * See file `support-issue-reproduction-v7.3mf` for the face that does not work
        * In this file, supports are produced as expected when I select "Supports: Everywhere"
            * This generates more supports than I want, but those supports do include the support for the face parallel to the build plate
    * [image]

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
* `support-issue-reproduction-v7.3mf` demonstrates the issue
* `support-issue-expected-v7.3mf` demonstrates the face that works as expected
[support-issue.zip](https://github.com/prusa3d/PrusaSlicer/files/7492618/support-issue.zip)

bubnikv, 2021-11-08:
Is the problem connected to paint-on supports?

There is a bug fix on master
b6c4e94d815dcb772be265134bf1e57e7beef6bd

    Fixing projection of bottom surfaces in MM segmentation and for
    support blockers / enforcers.

    All slicing functions shall produce consistent results with the same
mesh, same transformation matrix and slicing parameters.
    Namely, slice_mesh_slabs() shall produce consistent results with
slice_mesh() and slice_mesh_ex() in the sense, that projections made by
    slice_mesh_slabs() shall fall onto slicing planes produced by
slice_mesh().

    Before this commit, slice_mesh_slabs() projected bottom facing faces
upwards to its coplanar slicing plane,
    which is different from how slice_mesh() or slice_mesh_ex() work,
leading to ignored support enforcer / blocker strokes.

cuzelac, 2021-11-12:
Thanks for the response.

> Is the problem connected to paint-on supports?

From the description on the commit, it looks like it might fix my bug. I don't understand the slicer code well enough to dig in to the code to assess in more depth.

I'll check my case against beta2 when it arrives.

bubnikv, 2021-12-08:
Hopefully fixed, closing

closed 2021-12-08 by bubnikv