prusa3d/PrusaSlicer #7263: Project save CRASH

birdofplay, 2021-11-08:
### Version
_Version of PrusaSlicer used goes here_

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

lukasmatena, 2021-11-08:
@birdofplay Thanks. Can you please provide more info (any info, actually)?

birdofplay, 2021-11-08:
Thanks for taking the time to look into this issue.

The attached Screen grab files show my sequence.

Open Prusa slicer
Import stl
then,  without doing ANY THING ELSE
Save project as
CRASH

This occurs whether I use **Save Project** or **Save project as**

I have tried various STL files as well with similar results.

DOUBLE CLICK - OPEN PRUSA 
------------------------------------
[image]

Import Stl- 
------------------------------------
[image]

STL is an model of an F22 Nose 
------------------------------------
[image]

NOW  Without having done any operations , attempt to Save Project
----------------------------------------------------------------------
[image]

CRASH SCREEN
---------------------
[image]

Subsequent CRASH Message
-------------------------------
[image]

birdofplay, 2021-11-10:
I have Doweloaded the latest  2.4.0 beta1 both 64 and 32 bit 
And they BOTH CRASH  but even worse  they crash during slicing.

FidelCapo, 2021-11-10:
Can you please export the config bundle (File -> Export -> Export Config Bundle...), zip it and attach it with the STL file here? Thank you.

birdofplay, 2021-11-10:
I have exported a Config Bundle and and STL file  and sent it to HurkaF

thanks for looking into my situation.

Best

Bob

FidelCapo, 2021-11-11:
@birdofplay Unfortunately, I didn't get anything. You can upload the file to dropbox for example and put the link here.

birdofplay, 2021-11-11:
I have zipped the requested file into a single ~2Mb file and dragged it to this location
[Fuselage STL.zip](https://github.com/prusa3d/PrusaSlicer/files/7521056/Fuselage.STL.zip)

lukasmatena, 2021-11-22:
I reproduced the crash (a22bc7f7c8c2823a8c3bf2284bbf284376314da3):
```
=================================================================
==19354==ERROR: AddressSanitizer: heap-buffer-overflow on address 0x6020008bb498 at pc 0x0000042f7f1e bp 0x7ffcbdb3faf0 sp 0x7ffcbdb3fae0
READ of size 8 at 0x6020008bb498 thread T0 (slic3r_main)
    #0 0x42f7f1d in Slic3r::DoubleSlider::Control::draw_ruler(wxDC&) (/home/lukas/Slic3r/Slic3r/build_release/src/prusa-slicer+0x42f7f1d)
    #1 0x430e0da in Slic3r::DoubleSlider::Control::render() (/home/lukas/Slic3r/Slic3r/build_release/src/prusa-slicer+0x430e0da)
    #2 0x5c793e0 in wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&) (/home/lukas/Slic3r/Slic3r/build_release/src/prusa-slicer+0x5c793e0)
    #3 0x5c79715 in wxEvtHandler::SearchDynamicEventTable(wxEvent&) (/home/lukas/Slic3r/Slic3r/build_release/src/prusa-slicer+0x5c79715)
    (various wxWidgets and GTK functions)
    #72 0x5ad10c2 in wxWindow::Update() (/home/lukas/Slic3r/Slic3r/build_release/src/prusa-slicer+0x5ad10c2)
    #73 0x4237fda in Slic3r::GUI::Preview::update_layers_slider(std::vector<double, std::allocator<double> > const&, bool) (/home/lukas/Slic3r/Slic3r/build_release/src/prusa-slicer+0x4237fda)
    #74 0x423d466 in Slic3r::GUI::Preview::load_print_as_fff(bool) (/home/lukas/Slic3r/Slic3r/build_release/src/prusa-slicer+0x423d466)
    #75 0x424007a in Slic3r::GUI::Preview::load_print(bool) (/home/lukas/Slic3r/Slic3r/build_release/src/prusa-slicer+0x424007a)
    #76 0x2ef5d85 in Slic3r::GUI::Plater::priv::update_fff_scene() (/home/lukas/Slic3r/Slic3r/build_release/src/prusa-slicer+0x2ef5d85)
    #77 0x2f7b6d6 in Slic3r::GUI::Plater::priv::on_process_completed(Slic3r::SlicingProcessCompletedEvent&) (/home/lukas/Slic3r/Slic3r/build_release/src/prusa-slicer+0x2f7b6d6)
    
0x6020008bb498 is located 0 bytes to the right of 8-byte region [0x6020008bb490,0x6020008bb498)
allocated by thread T0 (slic3r_main) here:
    #0 0x7fe39db4a9df in operator new(unsigned long) (/usr/lib/x86_64-linux-gnu/libasan.so.5+0x10d9df)
    #1 0x42d79db in Slic3r::DoubleSlider::Control::Ruler::init(std::vector<double, std::allocator<double> > const&) (/home/lukas/Slic3r/Slic3r/build_release/src/prusa-slicer+0x42d79db)
    #2 0x4237e91 in Slic3r::GUI::Preview::update_layers_slider(std::vector<double, std::allocator<double> > const&, bool) (/home/lukas/Slic3r/Slic3r/build_release/src/prusa-slicer+0x4237e91)
    #3 0x423d466 in Slic3r::GUI::Preview::load_print_as_fff(bool) (/home/lukas/Slic3r/Slic3r/build_release/src/prusa-slicer+0x423d466)
    #4 0x424007a in Slic3r::GUI::Preview::load_print(bool) (/home/lukas/Slic3r/Slic3r/build_release/src/prusa-slicer+0x424007a)
    #5 0x2ef5d85 in Slic3r::GUI::Plater::priv::update_fff_scene() (/home/lukas/Slic3r/Slic3r/build_release/src/prusa-slicer+0x2ef5d85)

bubnikv, 2021-12-08:
fixed

birdofplay, 2021-12-08:
bubnikv:

Thank you for looking into my problem.

I am excited to try the latest **fixed version** of the legendary Prusa slicer.

Is it available for download yet ?  ( Win 7 64 bit ) 

If so, what version or revision number should I look for? 

Thank you very much,

Bob

bubnikv, 2021-12-08:
https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.4.0-beta3

birdofplay, 2021-12-08:
OK thanks for the reply.

But there is no mention of #7263 in the "FIXES" list. 
and unless I misunderstand the timing of reports and releases
the Beta3 was released 4 days ago. 

Regardless I WILL download Beta 3 and test it again.

bubnikv, 2021-12-08:
It is supposed to be fixed with https://github.com/prusa3d/PrusaSlicer/commit/a0f415a2d1d909de0e4f96ca5e2f172dc90b06eb
Sometimes we forget to mention a bug being fixed in a change log, I am sorry for that.

birdofplay, 2021-12-08:
I have tried both 2.33 and the more recent Beta3 unfortunately with no improvement.
If I attempt to **Save Project as  the Program Crashes
subsequently
If I just try to Slice either a Simple small STL or a Complicated Large part  the Program Crashes

I must conclude that my old Win7 machine is the problem.

bubnikv, 2021-12-08:
@birdofplay How much RAM do you have? You may be low on RAM.

I have attached a 3MF that I have saved off your profiles and STL with no issue.

[Fuz1.zip](https://github.com/prusa3d/PrusaSlicer/files/7678220/Fuz1.zip)

birdofplay, 2021-12-08:
16 GB

Intel q45/q43 onboard graphics chip set

[image]

bubnikv, 2021-12-08:
16GB RAM should be plenty for slicing.

birdofplay, 2021-12-08:
I just tried your 3mf on Beta3 and it Crashed at ~88% complete

birdofplay, 2021-12-08:
I just ried your 3mf fiel on 2.3.3 and whiel it warned againt certian setting MAY not be comaptable
It DID Slice without crash
[image]
.

lukasmatena, 2021-12-09:
@birdofplay Thanks. Could you try to run the beta3 from command line invoked as `prusa-slicer-console.exe --loglevel=5`? It will log into the console and might give a clue about where it crashes. Could you generate the log for both crashes (saving 3MF and exporting G-Code) and paste it here? Thanks.

It appears that it crashes when saving a file. Does the crash occur when you attempt to save to a different drive than before? Is there anything special about your setup, such as some obscure filesystem (i.e. different from NTFS), heavily throttled access rights and similar? It is possible that the very creating the file fails (but PrusaSlicer should of course handle that gracefully).

birdofplay, 2021-12-09:
Lucas, 

This machine is an HP E8400 with the only upgrade being 8GB ram NOW 16 GB ram
I uses the Onboard Intel Video chipset so NO "trick" Graphics cards.
It has an NTFS file system on the C: drive and ~ almost 200GB space left out of 1T
And of course Win7 and I have been accepting updates to it for years now.

It runs DOOM nicely ;>}
Alos runs Companion OPENTX for Transmitter programming of models.

It will run Cura 4.7 although very very slowly. 
Screen rotations are very jerky and Menu slider on Left is equally Jerky and slow

It will run S3D with no problems.

I runs Rhino 3d V5 a 3d CAD for my own designs.

I DO NOT run any of my printers with USB.
I Only use SD cads or thumb drives. 

To your reuest:

1. I created a shortcut with the arguments that you asked for and ran it.

2. I got a console window and copied that text to a File created in Textpad7.
3. Then I tried to Slice an stl  model and it crashed and the Console went away as well. ? !

So I couldn't capture any more of it's logging.

Is there a way to send logging to a specific place ?
Possibly with a redirection argument in the startup command string  ? 

If this is not enough you may need to help with the diagnostics.

Regardless, thank you for looking into this issue.
Best,
Bob

PS
The Zip file should have a file with the Screen cap of console at the top and later the
Prusaresearch..ini

However, shouldn't all of that also be IN the Config Bundle ?   

[PrusaSlicer.zip](https://github.com/prusa3d/PrusaSlicer/files/7687287/PrusaSlicer.zip)

birdofplay, 2021-12-09:
I went back in a watched the console as it Crashed during a slice of Benchy.
Benchy imported OK.
Then Prusa was good with an export of the Config bundle

Then On Slice at ~ 88% and the message Configuring Skirt and Brim .appeared   ... It crashed

BUT the Console WHILE NOW FROZEN was still visible.
It was reporting Pagefile max size items - See Pix below.

So I went in and **Increased my pagefile max** and rebooted.
My system seems a bit more "LAGGY" now.
but the same symptoms occur in Prusa Slicer - CRASH when slicing and saving Project.

[image]

bubnikv, 2021-12-11:
Frankly I would not wander if your old computer had hardware issues, for example memory (RAM) corruption issues. This happened to me recently. You may try to run for example

https://www.memtest86.com/#:~:text=MemTest86%20is%20the%20original%2C%20free%2C%20stand%20alone%20memory,a%20series%20of%20comprehensive%20algorithms%20and%20test%20patterns.

birdofplay, 2021-12-11:
That would certainly be an exhaustive mem test.
While I have DL'd it and will run it later, 
right now I have to wonder why I am NOT having any issues with 
CAD, Games, NETwork programs, E-Mail, my web site maintenance.

Simplify3D works quite nicely.
The older Cura 15 that came with my 3DPrinter and the New Cura 4.7
both Work with no similar **file save issues** and NO CRASHING.

So while running a mem chack will be interesting and potentially informative
I seriously doubt if it will reveal any salient issues.

lukasmatena, 2022-03-30:
This issue should be fixed on master for 2.5 release. I'm no longer able to reproduce it after @YuSanka's fix. Closing.

closed 2022-03-30 by lukasmatena