prusa3d/PrusaSlicer #7269: UI Request:  Object "hide" property - similar to "printable" property

clintonefisher, 2021-11-08:
### Version
2.3.3

### Operating system type + version
Win10 Pro

### 3D printer brand / version + firmware version (if known)
N/A - Prusa MK3S/MMU2S

### Behavior
New Feature Request:
Similar to the "printable" property in the object table, a "hide" property would hide the object in the 3D Editor View.  This would allow for a much tidier workspace and easier selection and manipulation of objects in a project containing many objects.

The icon for "printable" could change to a printer icon, and the hide icon could become the eye icon.
The two columns should probably be next to each other, perhaps highlighting red when a hidden object remains printable.

With objects hidden, it will be more important to add a right-click context menu to the object table entries so that right-clicking an object in the table behaves just like right-clicking it in the 3D Editor View.
(I will search for an existing feature request for the table context menu, and if not ,will submit a request)

P.S. Thank you to everyone out there who contributes to this project.  What an amazing creation!
I love seeing it get better and better as time progresses.  Thank you!!!

-Clint

lukasmatena, 2021-11-09:
Thanks. To be honest, I'm not sure if we need it. Let's wait for more opinions.

clintonefisher, 2021-11-11:
Is there an existing function that I do not know about to hide non-printable objects from the 3D Editor View?

That's all I really need.   And for me, it is definitely a need, not a want.  Printable objects in the 3D Editor View are often hidden by non-printable objects, which makes them much more difficult to select and manipulate.

Thank you for your consideration.

bubnikv, 2021-11-12:
Why do you want to hide a non-printable object? To place another object over it? I guess we don't understand your workflow.

clintonefisher, 2021-11-13:
> Why do you want to hide a non-printable object? To place another object over it? I guess we don't understand your workflow.

Yes, I have dozens of parts for a single project that I am printing separately.   I don't want to have to create a separate project file for each part, so I simply pile all the parts on the build plate in one project, and only set one or two parts as printable at a time.   I also move the parts around my print bed frequently to reduce the overall wear on any one spot, so there are many time parts in the 3D Editor View are overlapped and obscuring each other.   The only real problem with my workflow approach is that, without being able to hide items in the 3D Editor View, it makes resize/rotate operations, along with adding and editing modifiers, very tedious.

I realize my workflow might be unexpected and not something the original designers intended, but then again remember that celfone camera flash LEDs were not initially intended to be used as flashlights either.  It was only after people created hacks to switch the LED on that manufacturers realized that they had an unintended yet highly desirable feature in their phones.  

Take a look at almost any good design software - they allow hiding objects and/or layers.  Take Fusion 360 for example: You can hide objects from the editor view.  So, in PrusaSlicer, object hiding just makes sense as a desirable feature, and it would sure help my workflow in the way I work with my own projects.

Thank you for your time, consideration, and your work on PrusaSlicer.

lukasmatena, 2021-11-15:
@clintonefisher I also have a project with many parts and I solved the same issue by placing them outside the bed (where they are ignored), and only moving the ones I want to print on the bed. I am not sure if having many objects piled and overlapping on the build plate and switching them on and off would be equally easy to understand. Would this solve any of your issues?

clintonefisher, 2021-11-15:
> @clintonefisher I also have a project with many parts and I solved the same issue by placing them outside the bed (where they are ignored), and only moving the ones I want to print on the bed. I am [...]

Interesting workaround.  Thank you.   I'd honestly not considered dragging disabled objects outside the print bed just to get them out of the way.    I still think that the an option to hide the objects from the 3D Editor View would be very useful to many, but I also appreciate the (smack-my-head) simple workaround.  :)  I'll see how it goes.

Thank you!

clintonefisher, 2022-05-09:
> @clintonefisher I also have a project with many parts and I solved the same issue by placing them outside the bed (where they are ignored), and only moving the ones I want to print on the bed. I am [...]

Hi again @lukasmatena, since your suggestion to drag objects out of the print area to "hide" them, I've been doing so - thank you for the suggestion - it's certainly better than nothing.   After a while though, I still miss having a "Hide in Editor" column in the object list - mostly because position is often important to a layout, and it's a pain to move objects when I could just click a visibility icon on and off depending on which sets of parts I want to print.
So ya, if it were up to me, I'd still implement a "Hide" column, just to the left of the "Printable" column", using the eyeball icon for hide, and a printer icon for printable.

Just a suggestion.  :)  As always, thanks for all you do!

edwardhorsford, 2022-11-07:
I'd also really like this. I've got one large modifier at the top of my model, and this prevents be easily seeing / selecting stuff below. It would be so useful to be able to hide and show parts and modifiers to manipulate them.

Gurrrra, 2023-03-09:
How is this feature not implemented yet? It really is a no brainer. Sometimes you have many objects imported and generally want them placed in the middle for printing, so having to move them out of the way while also pressing the eye icon to disable them for printing is a bit of wonky way of working.
While I was searching for this I found the original request for having a checkbox for disabling an object for print AND one checkbox hiding it, and that request was closed after it was implemented, except they only implemented half the request. So please please implement the other half of that request now :)

primarchlion, 2023-06-14:
I am doing multicolor with a single extruder.  Printing text onto an object.  I print the text first and then the rest of the object goes on top of it.  It would be very nice to be able to hide objects to make sure that one model was in the proper place without moving the other one.

Same with "negative volumes".  if you create a negative volume inside an object, you'll never see it in the 3D editor view because you can't hide the primary object.

GithubUser99999999, 2023-06-28:
I would also love this feature. Right now with PrusaSlicer 2.6 it is simply impossible to position modifiers correctly inside an object since you just can't see it, because it's not possible to just hide the main print object.

kylek29, 2023-08-22:
Agreed on the visual aspect, this is something that I think has popped up more since the extra modifier flexibility in 2.6 .. ability to toggle visibility and printability of parts would be an excellent addition.

Another way we could aid the visual side of this is to have option toggles for shade w/outlines, and random component colors (assigns a random color (visual reference only) to each "part" within a model).

neophyl, 2023-08-22:
The visibility of modifiers inside other objects was a feature request I asked for ages ago back in 2020 https://github.com/prusa3d/PrusaSlicer/issues/3678

sg000, 2024-07-09:
Ditto here. Many objects in one project. Turn the visibility on and off as needed would be great. I have to move them on and off the build plate or have them all jammed on top of each other. Would be really nice to have!

closed 2026-09-29 by prusaslicer-bot[bot]