prusa3d/PrusaSlicer #7270: Feature Request: Right-click in object table should bring up context-menu - same as in 3D Editor View

clintonefisher, 2021-11-08:
### Version
2.3.3

### Operating system type + version
Win10 Pro

### 3D printer brand / version + firmware version (if known)
N/A (Prusa MK3S/MMU2S)

### Behavior
New Feature Request:

When right-clicking an object in the object table, there is currently no action other than selecting the object.

It seems reasonable to have a right-click on an object in the object table bring up the same context-menu as a right-click in the 3D Editor View would.   

Left-click behavior in the object table would remain the same; simply selecting the object.

As always, thank you to everyone working on this amazing project!!!

-Clint

lukasmatena, 2021-11-09:
@clintonefisher
The menu is actually there, but you must right-click the icon in the rightmost column. It does not open when clicking the line itself.
[image]
As far as I know, we ran into some limitations with wxWidgets and we worked around them this way.
@YuSanka Does it have to stay this way or is there something we can do?

YuSanka, 2021-11-19:
There are no limitations in this case. THe context menu is shown just for "Editing" column. This was done by design. But we can change that. With a [commit](https://github.com/prusa3d/PrusaSlicer/commit/172b97cc2a3ed1290ee0c02b164379d0e38bc39e), I also allow the context menu to be shown for the "Name" column.

FidelCapo, 2021-11-25:
This issue should be fixed in the next release.
Closing.

closed 2021-11-25 by FidelCapo

foreachthing, 2021-11-26:
I love it! :sparkling_heart:
Question: is the _Edit_ column still necessary?

clintonefisher, 2021-12-21:
> This issue should be fixed in the next release. Closing.

Thank you FidelCapo and YuSanka!!   
Much love and gratitude to the whole PrusaSlicer team and all the contributors!

clintonefisher, 2021-12-30:
I just wanted to come back here and say that every time I right click on this and get a menu it makes me so happy, I get giddy.  You are all incredibly awesome.  Thank you all again and have a very Happy 2022!  <3 <3 <3