prusa3d/PrusaSlicer #7272: Single extruder multimaterial has stopped working

Hades32, 2021-11-08:
### Version
2.3.3 and 2.4.0 Beta 1

### Operating system type + version
Windows 11

### 3D printer brand / version + firmware version (if known)
Custom: [PrusaSlicer_config_bundle.zip](https://github.com/prusa3d/PrusaSlicer/files/7500148/PrusaSlicer_config_bundle.zip)

### Behavior
* load config above
* create a simple cylinder and add a color change in the middle from extruder 0 to 1
* _Expected Results_
  * GCode should container commands for cooling moves and retraction etc
* _Actual Results_
  * tool-change code is created but no SEMM related commands
[example semm bug.zip](https://github.com/prusa3d/PrusaSlicer/files/7500168/example.semm.bug.zip)

_Is this a new feature request?_
No, bug report. This worked before, but I'm not sure in which version

lukasmatena, 2021-11-09:
@Hades32 The SEMM stuff is only used when the wipe tower is enabled (Print Settings->Multiple Extruders). You have it disabled.

Hades32, 2021-11-09:
Thanks @lukasmatena! But why is that? Sometimes I don't need perfect color changes...

lukasmatena, 2021-11-09:
Because it is currently implemented this way, because of historical reasons. It would probably be possible to separate the toolchange from the actual wipe tower, but it would take some time. I'll not promise we'll do it, but I can rename this issue and keep it open to track this request, if that's fine.

Hades32, 2021-11-09:
For folks coming here after me, this is the workaround I settled with. Use this as custom GCode for tool changes
```gcode
;TOOL_CHANGE from {previous_extruder} to {next_extruder}

{if previous_extruder != next_extruder && previous_extruder != -1}
; get filament out of the way
G1 E-150
; do some up down movements to cool filament without sticking to the tube
G1 E10
G1 E-10
G1 E10
G1 E-10
G1 E10
G1 E-10
G1 E10
G1 E-10
; switch tool
T1
; get back previously stored filament
G1 E150
; press it good in
G1 E3
{else}
; nothing to do
{endif}
```

closed 2026-09-29 by prusaslicer-bot[bot]