prusa3d/PrusaSlicer #7273: PrusSlicer app icon

double3, 2021-11-08:
### Version
2.3.3+x64

### Operating system type + version
macOS Monterey 12.0.1

### 3D printer brand / version + firmware version (if known)
MK3S+

Can you please update the PrusaSlicer app icon so it matches the default styling of native MacOS apps? It's currently the only app on my dock that looks out of place. Not a huge deal, obviously, but my OCD says it's super important. :)

[image]

JohnOCFII, 2021-11-09:
The icon has already been updated in the Beta release for the upcoming 2.4.

lukasmatena, 2021-11-09:
@JohnOCFII Thanks, that's right. Closing

closed 2021-11-09 by lukasmatena