prusa3d/PrusaSlicer #7274: Segfault when clicking "Variable Layer Height" button

BruceJL, 2021-11-08:
### Version
Version 2.4.0-beta1+linux-x64

### Operating system type + version
_What OS are you using, and state any version _
Debian Testing

_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_
```
PrusaSlicer
Version:   2.4.0-beta1+linux-x64
Build:     PrusaSlicer-2.4.0-beta1+linux-x64-202110251331

Operating System:    Unix
System Architecture: 64 bit
System Version:      Linux 5.14.0-2-amd64 x86_64
Total RAM size [MB]: 8,194MB
OpenGL installation
GL version:   3.0 Mesa 21.2.5
Vendor:       Intel Open Source Technology Center
Renderer:     Mesa DRI Intel(R) HD Graphics 4000 (IVB GT2)
GLSL version: 1.30
<details>
<summary>Installed extensions:</summary>
GL_3DFX_texture_compression_FXT1
GL_AMD_conservative_depth
GL_AMD_draw_buffers_blend
GL_AMD_multi_draw_indirect
GL_AMD_seamless_cubemap_per_texture
GL_AMD_shader_trinary_minmax
GL_AMD_texture_texture4
GL_ANGLE_texture_compression_dxt3
GL_ANGLE_texture_compression_dxt5
GL_APPLE_object_purgeable
GL_APPLE_packed_pixels
GL_ARB_ES2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_compute_variable_group_size
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_framebuffer_no_attachments
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_get_program_binary
GL_ARB_get_texture_sub_image
GL_ARB_gl_spirv
GL_ARB_half_float_pixel
GL_ARB_half_float_vertex
GL_ARB_indirect_parameters
GL_ARB_instanced_arrays
GL_ARB_internalformat_query
GL_ARB_internalformat_query2
GL_ARB_invalidate_subdata
GL_ARB_map_buffer_alignment
GL_ARB_map_buffer_range
GL_ARB_multi_bind
GL_ARB_multi_draw_indirect
GL_ARB_multisample
GL_ARB_multitexture
GL_ARB_occlusion_query
GL_ARB_occlusion_query2
GL_ARB_parallel_shader_compile
GL_ARB_pipeline_statistics_query
GL_ARB_pixel_buffer_object
GL_ARB_point_parameters
GL_ARB_point_sprite
GL_ARB_polygon_offset_clamp
GL_ARB_program_interface_query
GL_ARB_provoking_vertex
GL_ARB_robustness
GL_ARB_sample_shading
GL_ARB_sampler_objects
GL_ARB_seamless_cube_map
GL_ARB_seamless_cubemap_per_texture
GL_ARB_separate_shader_objects
GL_ARB_shader_atomic_counter_ops
GL_ARB_shader_atomic_counters
GL_ARB_shader_bit_encoding
GL_ARB_shader_clock
GL_ARB_shader_draw_parameters
GL_ARB_shader_group_vote
GL_ARB_shader_image_load_store
GL_ARB_shader_image_size
GL_ARB_shader_objects
GL_ARB_shader_precision
GL_ARB_shader_storage_buffer_object
GL_ARB_shader_texture_image_samples
GL_ARB_shader_texture_lod
GL_ARB_shading_language_100
GL_ARB_shading_language_420pack
GL_ARB_shading_language_include
GL_ARB_shading_language_packing
GL_ARB_shadow
GL_ARB_spirv_extensions
GL_ARB_sync
GL_ARB_texture_barrier
GL_ARB_texture_border_clamp
GL_ARB_texture_compression
GL_ARB_texture_compression_bptc
GL_ARB_texture_compression_rgtc
GL_ARB_texture_cube_map
GL_ARB_texture_cube_map_array
GL_ARB_texture_env_add
GL_ARB_texture_env_combine
GL_ARB_texture_env_crossbar
GL_ARB_texture_env_dot3
GL_ARB_texture_filter_anisotropic
GL_ARB_texture_float
GL_ARB_texture_gather
GL_ARB_texture_mirror_clamp_to_edge
GL_ARB_texture_mirrored_repeat
GL_ARB_texture_multisample
GL_ARB_texture_non_power_of_two
GL_ARB_texture_query_levels
GL_ARB_texture_query_lod
GL_ARB_texture_rectangle
GL_ARB_texture_rg
GL_ARB_texture_rgb10_a2ui
GL_ARB_texture_storage
GL_ARB_texture_storage_multisample
GL_ARB_texture_swizzle
GL_ARB_texture_view
GL_ARB_timer_query
GL_ARB_transform_feedback2
GL_ARB_transform_feedback3
GL_ARB_transform_feedback_instanced
GL_ARB_transform_feedback_overflow_query
GL_ARB_transpose_matrix
GL_ARB_uniform_buffer_object
GL_ARB_vertex_array_bgra
GL_ARB_vertex_array_object
GL_ARB_vertex_attrib_binding
GL_ARB_vertex_buffer_object
GL_ARB_vertex_program
GL_ARB_vertex_shader
GL_ARB_vertex_type_10f_11f_11f_rev
GL_ARB_vertex_type_2_10_10_10_rev
GL_ARB_window_pos
GL_ATI_blend_equation_separate
GL_ATI_draw_buffers
GL_ATI_separate_stencil
GL_ATI_texture_env_combine3
GL_ATI_texture_float
GL_EXT_EGL_image_storage
GL_EXT_EGL_sync
GL_EXT_abgr
GL_EXT_bgra
GL_EXT_blend_color
GL_EXT_blend_equation_separate
GL_EXT_blend_func_separate
GL_EXT_blend_minmax
GL_EXT_blend_subtract
GL_EXT_compiled_vertex_array
GL_EXT_copy_texture
GL_EXT_demote_to_helper_invocation
GL_EXT_direct_state_access
GL_EXT_draw_buffers2
GL_EXT_draw_instanced
GL_EXT_draw_range_elements
GL_EXT_fog_coord
GL_EXT_framebuffer_blit
GL_EXT_framebuffer_multisample
GL_EXT_framebuffer_multisample_blit_scaled
GL_EXT_framebuffer_object
GL_EXT_framebuffer_sRGB
GL_EXT_gpu_program_parameters
GL_EXT_gpu_shader4
GL_EXT_memory_object
GL_EXT_memory_object_fd
GL_EXT_multi_draw_arrays
GL_EXT_packed_depth_stencil
GL_EXT_packed_float
GL_EXT_packed_pixels
GL_EXT_pixel_buffer_object
GL_EXT_point_parameters
GL_EXT_polygon_offset_clamp
GL_EXT_provoking_vertex
GL_EXT_rescale_normal
GL_EXT_secondary_color
GL_EXT_semaphore
GL_EXT_semaphore_fd
GL_EXT_separate_specular_color
GL_EXT_shader_framebuffer_fetch_non_coherent
GL_EXT_shader_integer_mix
GL_EXT_shader_samples_identical
GL_EXT_shadow_funcs
GL_EXT_stencil_two_side
GL_EXT_stencil_wrap
GL_EXT_subtexture
GL_EXT_texture
GL_EXT_texture3D
GL_EXT_texture_array
GL_EXT_texture_compression_dxt1
GL_EXT_texture_compression_rgtc
GL_EXT_texture_compression_s3tc
GL_EXT_texture_cube_map
GL_EXT_texture_edge_clamp
GL_EXT_texture_env_add
GL_EXT_texture_env_combine
GL_EXT_texture_env_dot3
GL_EXT_texture_filter_anisotropic
GL_EXT_texture_integer
GL_EXT_texture_lod_bias
GL_EXT_texture_object
GL_EXT_texture_rectangle
GL_EXT_texture_sRGB
GL_EXT_texture_sRGB_R8
GL_EXT_texture_sRGB_decode
GL_EXT_texture_shadow_lod
GL_EXT_texture_shared_exponent
GL_EXT_texture_snorm
GL_EXT_texture_swizzle
GL_EXT_timer_query
GL_EXT_transform_feedback
GL_EXT_vertex_array
GL_EXT_vertex_array_bgra
GL_IBM_multimode_draw_arrays
GL_IBM_rasterpos_clip
GL_IBM_texture_mirrored_repeat
GL_INGR_blend_func_separate
GL_INTEL_blackhole_render
GL_INTEL_performance_query
GL_KHR_blend_equation_advanced
GL_KHR_context_flush_control
GL_KHR_debug
GL_KHR_no_error
GL_KHR_parallel_shader_compile
GL_KHR_robustness
GL_MESA_pack_invert
GL_MESA_shader_integer_functions
GL_MESA_texture_signed_rgba
GL_MESA_window_pos
GL_NV_blend_square
GL_NV_compute_shader_derivatives
GL_NV_conditional_render
GL_NV_depth_clamp
GL_NV_fog_distance
GL_NV_half_float
GL_NV_light_max_exponent
GL_NV_packed_depth_stencil
GL_NV_primitive_restart
GL_NV_texgen_reflection
GL_NV_texture_barrier
GL_NV_texture_env_combine4
GL_NV_texture_rectangle
GL_OES_EGL_image
GL_OES_read_format
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_border_clamp
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod
GL_SUN_multi_draw_arrays
</details>
```

### 3D printer brand / version + firmware version (if known)
MK3S with Bondtech hotend - 3.10.0

### Behavior
_Describe the problem_
PrusaSlicer will segfault when clicking the "Variable Layer Height" button.

_Steps needed to reproduce the problem_
* Open any model and click on the "Variable Layer Height" button 
* PrusaSlicer crashes.

_If this is a command-line slicing issue, include the options used_
No command line options used.

_Expected Results_
The variable layer height tool works.

_Actual Results_
PrusaSlicer segfaults

_Screenshots from __*PrusaSlicer*__ preview are preferred_
None to supply

_Is this a new feature request?_
No.

#### Project File (.3MF) where problem occurs
Example file uploaded
[weight bracket.zip](https://github.com/prusa3d/PrusaSlicer/files/7500794/weight.bracket.zip)

BruceJL, 2021-11-08:
Here's a --loglevel 5 of the segfault.

```
./PrusaSlicer-2.4.0-beta1+linux-x64-202110251331.AppImage --loglevel 5
[2021-11-08 17:17:39.767723] [0x00007f17b72298c0] [trace]   Initializing StaticPrintConfigs
[2021-11-08 17:17:39.875407] [0x00007f17b72298c0] [debug]   full path: 4127782352887830672
[2021-11-08 17:17:39.875445] [0x00007f17b72298c0] [debug]   single instance: undefined. other params: /tmp/.mount_PrusaSi2ZNa7/usr/bin/bin/prusa-slicer;--loglevel;5
[2021-11-08 17:17:39.875458] [0x00007f17b72298c0] [debug]   full lock path: /home/bruce/.config/PrusaSlicer-beta/cache/4127782352887830672.lock
[2021-11-08 17:17:39.875507] [0x00007f17b72298c0] [debug]   Creating lockfile.
[2021-11-08 17:17:39.875540] [0x00007f17b72298c0] [info]    Instance check: Another instance not found or single-instance not set.
[2021-11-08 17:17:39.943234] [0x00007f17b72298c0] [trace]   System language detected (user locales and such): en_CA
[2021-11-08 17:17:39.945639] [0x00007f17b72298c0] [trace]   Best translation language detected (may be different from user locales): en_GB
[2021-11-08 17:17:39.945673] [0x00007f17b72298c0] [trace]   Switching wxLocales to en_GB
[2021-11-08 17:17:40.076466] [0x00007f17b72298c0] [info]    Checking if indices need to be installed from resources...
[2021-11-08 17:17:40.581924] [0x00007f17b72298c0] [info]    Undo / Redo snapshot taken: New Project, Undo / Redo stack memory: 0MB Resident memory: 80MB; Shared memory: 48MB; Private memory: 32MB; Peak memory usage: 80MB
[2021-11-08 17:17:40.583149] [0x00007f17a5cff640] [trace]   Dbus object /com/prusa3d/prusaslicer/InstanceCheck/Object4127782352887830672 registered. Starting listening for messages.
[2021-11-08 17:17:41.102760] [0x00007f17b72298c0] [debug]   TriangleMesh::repair() started
[2021-11-08 17:17:41.103685] [0x00007f17b72298c0] [debug]   TriangleMesh::repair() finished
[2021-11-08 17:17:42.198885] [0x00007f17b72298c0] [info]    Checking for cached configuration updates...
[2021-11-08 17:17:42.198936] [0x00007f17b72298c0] [info]    Confing bundle not installed for vendor FLSun, skipping: 
[2021-11-08 17:17:42.198961] [0x00007f17b72298c0] [info]    Confing bundle not installed for vendor Creality, skipping: 
[2021-11-08 17:17:42.208128] [0x00007f17b72298c0] [debug]   Vendor: Prusa Research, version installed: 1.2.7, version cached: 1.4.0-beta1
[2021-11-08 17:17:42.222184] [0x00007f17b72298c0] [debug]   PresetUpdater: Copying "/home/bruce/.config/PrusaSlicer-beta/cache/PrusaResearch.idx" -> "/home/bruce/.config/PrusaSlicer-beta/vendor/PrusaResearch.idx"
[2021-11-08 17:17:42.222240] [0x00007f17b72298c0] [debug]   boost::filesystem::permisions before copy error message (this could be irrelevant message based on file system): No such file or directory
[2021-11-08 17:17:42.224692] [0x00007f17b72298c0] [info]    Confing bundle not installed for vendor Anycubic, skipping: 
[2021-11-08 17:17:42.224714] [0x00007f17b72298c0] [info]    Confing bundle not installed for vendor BIBO, skipping: 
[2021-11-08 17:17:42.224735] [0x00007f17b72298c0] [info]    Confing bundle not installed for vendor LulzBot, skipping: 
[2021-11-08 17:17:42.224755] [0x00007f17b72298c0] [info]    Confing bundle not installed for vendor TriLAB, skipping: 
[2021-11-08 17:17:42.224773] [0x00007f17b72298c0] [info]    Confing bundle not installed for vendor Artillery, skipping: 
[2021-11-08 17:17:42.224792] [0x00007f17b72298c0] [info]    Confing bundle not installed for vendor gCreate, skipping: 
[2021-11-08 17:17:42.224810] [0x00007f17b72298c0] [info]    Confing bundle not installed for vendor INAT, skipping: 
[2021-11-08 17:17:42.237136] [0x00007f179351f640] [info]    Downloading PrusaSlicer online version from: `https://files.prusa3d.com/wp-content/uploads/repository/PrusaSlicer-settings-master/live/PrusaSlicer.version2`
*   Trying 185.115.1.124:443...
* Connected to files.prusa3d.com (185.115.1.124) port 443 (#0)
* ALPN, offering http/1.1
* SSL connection using TLSv1.2 / ECDHE-RSA-AES256-GCM-SHA384
* ALPN, server accepted to use http/1.1
* Server certificate:
*  subject: CN=files.prusa3d.com
*  start date: Sep 10 19:03:45 2021 GMT
*  expire date: Dec  9 19:03:44 2021 GMT
*  subjectAltName: host "files.prusa3d.com" matched cert's "files.prusa3d.com"
*  issuer: C=US; O=Let's Encrypt; CN=R3
*  SSL certificate verify ok.
> GET /wp-content/uploads/repository/PrusaSlicer-settings-master/live/PrusaSlicer.version2 HTTP/1.1
Host: files.prusa3d.com
User-Agent: PrusaSlicer/2.4.0-beta1+linux-x64
Accept: */*

* Mark bundle as not supporting multiuse
< HTTP/1.1 200 OK
< Server: nginx
< Date: Mon, 08 Nov 2021 23:17:42 GMT
< Content-Length: 6
< Connection: keep-alive
< Last-Modified: Mon, 08 Nov 2021 22:59:57 GMT
< ETag: "6-5d04ef3a31826"
< Accept-Ranges: bytes
< 
* Connection #0 to host files.prusa3d.com left intact
[2021-11-08 17:17:42.865114] [0x00007f179351f640] [info]    Got PrusaSlicer online version: `2.3.3`. Sending to GUI thread...
[2021-11-08 17:17:42.868488] [0x00007f179351f640] [info]    Syncing configuration cache
[2021-11-08 17:17:42.868538] [0x00007f179351f640] [warning] No such vendor: FLSun
[2021-11-08 17:17:42.868573] [0x00007f179351f640] [warning] No such vendor: Creality
[2021-11-08 17:17:42.868595] [0x00007f179351f640] [info]    Downloading index for vendor: Prusa Research
[2021-11-08 17:17:42.868655] [0x00007f179351f640] [info]    Get: `https://files.prusa3d.com/wp-content/uploads/repository/PrusaSlicer-settings-master/live/PrusaResearch//index.idx`
[... cut ...]
[2021-11-08 17:17:44.748596] [0x00007f179351f640] [warning] No such vendor: Artillery
[2021-11-08 17:17:44.748613] [0x00007f179351f640] [warning] No such vendor: gCreate
[2021-11-08 17:17:44.748632] [0x00007f179351f640] [warning] No such vendor: INAT
[2021-11-08 17:17:46.090055] [0x00007f17b72298c0] [info]    Undo / Redo snapshot taken: Load Project: weight bracket, Undo / Redo stack memory: 0MB Resident memory: 570MB; Shared memory: 67MB; Private memory: 503MB; Peak memory usage: 631MB
Segmentation fault
```

lukasmatena, 2021-11-10:
I cannot reproduce it on Linux, it is quite likely OpenGL related. I would suggest you try upgrading the graphics driver and MESA, but MESA is already the latest so I assume you're up-to-date. See #6846 and #6677 for other things to try and let us know how it went (if you want to).

Did it work before? I mean before upgrading to 2.4.0-beta1 or with earlier MESA/graphics drivers? Thanks.

lukasmatena, 2021-11-10:
Also, you may try running `./prusa-slicer --sw-renderer`, which turns off hardware acceleration. The performance will be affected, but it might give a clue.

BruceJL, 2021-11-11:
Running the appimage with the supplied command line does not launch Prusaslicer and yields the following:

./PrusaSlicer-2.4.0-beta1+linux-x64-202110251331.AppImage --sw-renderer
[2021-11-10 20:00:51.468309] [0x00007f8a984708c0] [trace]   Initializing StaticPrintConfigs
Unknown option --sw-renderer

Pretty sure the latest beta and latest stable (2.3.0) stopped working after the upgrade from Debian 11 to 12(testing). It's likely something to do with that upgrade.

I'm sorry, I couldn't see anything actionable in the linked tickets. I'm happy to try any other suggestions and report back.

Thank you for your input.

bubnikv, 2021-11-12:
We implemented the --sw-renderer switch for Windows only, where we load a Mesa SW renderer DLL.

On Linux, you have to enable software MESA or whatever it takes. You will likely find an answer in some other pending PrusaSlicer issue, unfortunately OpenGL on Linux is often wonky.

m-sundman, 2021-11-24:
Yes, it works if you run it like this:
`LIBGL_ALWAYS_SOFTWARE=1 ./PrusaSlicer-2.4.0-beta1+linux-x64-202110251331.AppImage`

enricoturri1966, 2022-04-14:
@stillpointx, @BruceJL

Could you please test if any of the following experimental builds solves the issue ?

Using OpenGL 3.3 core profile:

[](https://prusaslicer:slicer@slicerbuilds.prusa3d.com/files/opengl/PrusaSlicer-2.5.0-alpha0+395-et-opengl-3-linux-x64-GTK3-g095a765-202204130920.AppImage)
[](https://prusaslicer:slicer@slicerbuilds.prusa3d.com/files/opengl/PrusaSlicer-2.5.0-alpha0+395-et-opengl-3-linux-x64-GTK3-g095a765-202204130920.tar.bz2)

Using OpenGL ES 2.0:
[](https://prusaslicer:slicer@slicerbuilds.prusa3d.com/files/opengl/PrusaSlicer-2.5.0-alpha0+418-et-opengl-es-linux-x64-GTK3-gf1adfa5-202204131022.AppImage)
[](https://prusaslicer:slicer@slicerbuilds.prusa3d.com/files/opengl/PrusaSlicer-2.5.0-alpha0+418-et-opengl-es-linux-x64-GTK3-gf1adfa5-202204131022.tar.bz2)

Current master:
[](https://prusaslicer:slicer@slicerbuilds.prusa3d.com/files/opengl/PrusaSlicer-2.5.0-alpha0+363-linux-x64-GTK3-g09512c0-202204132308.AppImage)
[](https://prusaslicer:slicer@slicerbuilds.prusa3d.com/files/opengl/PrusaSlicer-2.5.0-alpha0+363-linux-x64-GTK3-g09512c0-202204132308.tar.bz2)

m-sundman, 2022-04-14:
> Could you please test if any of the following experimental builds solves the issue ?
> 
> Using OpenGL 3.3 core profile

Works.

> Using OpenGL ES 2.0

Works.

> Current master

Crashes.

BruceJL, 2022-04-16:
OpenGL 3.3: Works
OpenGl ES 2.0: Works
Current Master: Crashes

WetdesignSee, 2022-04-17:
I had the same issue - Crash on variable layer height. 
LIBGL_ALWAYS_SOFTWARE=1 ./PrusaSlicer-2.4.2-rc2+linux-x64-GTK2-202204131344.AppImage   *Worked
./PrusaSlicer-2.5.0-alpha0+395-et-opengl-3-linux-x64-GTK3-g095a765-202204130920.AppImage    *Worked
 Using Ubuntu 20.04.4 LTS

kubispe1, 2022-12-13:
Thanks to the community for the tests. OpenGL 3.3 looks stable and it is fixed in 2.6.0.
So, I close this issue.

closed 2022-12-13 by kubispe1