prusa3d/PrusaSlicer #7275: PrusaSlicer 2.2.0 never pushed software updates to me

rsfadia, 2021-11-08:
### Version
_2.2.0+

### Operating system type + version
_MacOS Monterey_

### Behavior
I just wanted to give a heads up to the Prusa team that my PrusaSlicer software did not notify my of a software update, even after I press "check for updates" in the configuration menu. I've been running macOS Big Sur (and recently updated to Monterey) if that is useful information. When I press "About PrusaSlicer", it says version 2.2.0+ at the top. I found out there was an update when I was reading about the painting on supports feature currently on version 2.3. Unfortunately I just updated to 2.3 so I can't investigate further into the problem, but thought you would want to know in case there are other folks living in the PrusaSlicer Stone Age (its been two days and I already love the ironing feature!). 

- Roy

lukasmatena, 2021-11-09:
The 'Check for updates' only checks for configuration updates, not for application updates. If you want to be notified about application updates, you must enable 'Check for application updates' in Preferences (I believe it was always on by default, though). It should then show a notification the first time it sees a new application version online. Once you confirm the notification, it does not appear again.

Is it possible that you either have that option off, or dismissed the notification when you saw it?

rsfadia, 2021-11-09:
I wouldn’t do that purposefully, and I wish I had held onto 2.2 a little longer so that I could check. However, me unchecking that box accidentally seems to be the most likely solution. I’m fond of updating software, so I wouldn’t ignored multiple update notifications. 

Thank you for the explanation!

kocikdav, 2021-11-10:
@lukasmatena is right. Also in version 2.3.0 we had crashing bug in notification about new releases. Thus we had to stop notifying about releases in versions 2.3.0 and older. It should be working in all newer versions though. 
In 2.4.0 there will be notification about alpha and beta releases too :)

closed 2021-11-10 by kocikdav