prusa3d/PrusaSlicer #7276: Slicer crashes after sending file to the printer.

cmoiceanu, 2021-11-09:
### Version
2.4.0-beta1+win64

Build: PrusaSlicer-2.4.0-beta1+win64-202110251333

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
Operating System: Windows
System Architecture: 64 bit
Windows Version: Windows 10 (build 19041), 64-bit edition

### 3D printer brand / version + firmware version (if known)
MK3S with LGX kit

### Behavior
Prusa MK3S configured with OctaPrint via a RaspeberyPI. 
Sending the file to the printer works the first time around, sending the file a second time crashes the app. The slicer app closes down unexpectedly. This worked fine in previous versions, and this is new and unexpected to the beta. The file doesn't matter, it can be any STL, works first time crashes the second time around. 

* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while 
[image]
all other files need to be zipped first (.zip, .gz)_

napter, 2021-11-09:
I've experienced this with the beta a few times. I was using the alphas before this and these didn't happen - these are new crashes and the beta feels less stable than the alphas.

lukasmatena, 2021-11-09:
Duplicate of #7170, fixed, closing.

closed 2021-11-09 by lukasmatena