prusa3d/PrusaSlicer #7277: 3Dconnextion does not have all axis available.

kareardo, 2021-11-09:
### Version
PrusaSlicer
Version:   2.3.3+win64
Build:     PrusaSlicer-2.3.3+win64-202107161027

### Operating system type + version
Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 19043), 64-bit edition
Total RAM size [MB]: 68,429MB
OpenGL installation
GL version:   4.6.0 NVIDIA 462.21
Vendor:       NVIDIA Corporation
Renderer:     GeForce RTX 3060 Laptop GPU/PCIe/SSE2
GLSL version: 4.60 NVIDIA

### Behavior
* _Describe the problem_
The 3Dconnexion space mouse no longer works in all dimensions.  It only works in a tilt of an axis.  It just tilts in the x-y diagonal direction.  no specific x or y direction.  No other dimensions work.

#### Project File (.3MF) where problem occurs
This happens on all files, an also when just started without any file loaded.

bubnikv, 2021-11-09:
What model? What drivers? Did you try 3DConnextion support?

szafran81, 2021-11-28:
+1 here.
Just stopped working some time ago.
Don't really know if it was after PS update or after 3DxWare update (other 3D apps work as they should).
My PS is v2.3.3 and 3DxWare is v10.8.3.3389

kareardo, 2021-11-29:
3dxware 10.8.3.3389
3dxwincore 17.8.3.18895
spacemouse wireless FW:4.41 / PID:A4321064927

It works just fine in Fusion 360.

bubnikv, 2021-11-29:
Did you guys try to call 3DConnexion support? PrusaSlicer is supported officially by the 3DConnexion drivers for some time already, the driver contains configuration files specific to PrusaSlicer.

kareardo, 2021-11-30:
I thought this was the support vector.  Oh well.

bubnikv, 2021-11-30:
@kareardo 

> I thought this was the support vector. Oh well.

We have so many issues caused by incorrect support of 3DConnexion of their products. They are charging a huge amount of money for their devices, they obsolete their super expensive devices very early. I think it is fair that they will support them.

bubnikv, 2021-12-22:
Please check https://www.help.prusa3d.com/en/article/3dconnexion-devices_1511
and the comments below.

The 3DConnexion drivers are supposed to install configuration file specific for PrusaSlicer:
https://github.com/prusa3d/PrusaSlicer/issues/4104#issuecomment-707727500

I see in the comments below https://www.help.prusa3d.com/en/article/3dconnexion-devices_1511
that maybe the newest 3DConnexion drivers don't install C:\Program Files\3Dconnexion\3DxWare\3DxWinCore\Cfg\prusa-slicer.xml
Would you please confirm that? If it is the case, then we have to notify 3DConnexion to fix their drivers.

szafran81, 2021-12-22:
I've just updated 3DXWare to 10.8.4 and the config file is present, but PS still behaves the same for me (rotates one axis only).

bubnikv, 2021-12-22:
I am sorry, but the best thing I can do is to recommend contacting
3DConnexion support. We cannot buy and support all of their products with
all of their drivers.

bubnikv, 2021-12-23:
I am not sure whether this helps and why.
https://github.com/prusa3d/PrusaSlicer/issues/7549#issuecomment-999678408

bubnikv, 2021-12-23:
To be up to date, I downloaded 3DxWare64_v10-8-4_r3407.exe
Deinstalled the old drivers, rebooted, removed whatever was left from Program Files from 3DConnexion driver (there was the old prusa-slicer.xml likely from the time we worked together with 3DConnexion on PrusaSlicer support being integrated into their drivers). Then I installed the driver mentioned above.

Now when I activate PrusaSlicer, the 3DConnexion settings shows 
[image]

szafran81, 2021-12-23:
> I am not sure whether this helps and why. [#7549 (comment)](https://github.com/prusa3d/PrusaSlicer/issues/7549#issuecomment-999678408)

Worked for me (don't forget to turn off the app while removing the file and then start it back again).

kareardo, 2021-12-27:
Solved it.  I just renamed the xml and now everything works!

---K

enricoturri1966, 2022-01-06:
Driver: 3DxWare64_v10-8-4_r3407.exe

I was able to get my SpaceMouse Pro Wireless working with PrusaSlicer by editing the file C:\Program Files\3Dconnexion\3DxWare\3DxWinCore\Cfg\prusa-slicer.xml
using C:\Program Files\3Dconnexion\3DxWare\3DxWinCore\Cfg\Blender.xml as reference.

This is the edited file:

```
<?xml version="1.0" encoding="utf-8"?>
<!--
(c) Copyright 3Dconnexion. All rights reserved. 
$Revision: 16358 $
-->
<AppCfg  xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" Default="true" CfgFormatVersion="1.3" ThisFileVersion="1.0">
  <CfgProperties>
    <ID>ID_PrusaSlicer</ID>
    <Name>PrusaSlicer</Name>
    <InheritsFromID>ID_Default_Cfg</InheritsFromID>
  </CfgProperties>
  <AppInfo>
    <Signature>
      <Name>PrusaSlicer</Name>
      <ExecutableName>prusa-slicer.exe</ExecutableName>
      <Transport>RawInput</Transport>
    </Signature>
  </AppInfo>
  <AuthorInfo>
    <DriverVersion>17.6.3.16588</DriverVersion>
    <Created>
      <Tool>3DxService.exe</Tool>
      <Version>17.6.3.16588</Version>
      <Date>2020:4:3:1:10:16:46:953</Date>
      <User>prusa</User>
    </Created>
    <LastChange>
      <Tool>3DxSmartUI.exe</Tool>
      <Version>1.2.19261.16572</Version>
      <Date>4/1/2020 10:22:47 AM UTC</Date>
      <User>prusa</User>
    </LastChange>
  </AuthorInfo>
</AppCfg>
```

bubnikv, 2022-01-06:
and we have reported to 3DConnexion

beeb, 2022-01-20:
Confirmed that editing the XML file and editing the `<InheritsFromID>` key to value `ID_Default_Cfg` fixes the problem.

bubnikv, 2022-01-27:
3DConnexion confirmed the issue with their driver configuration and they promised to fix it with their upcoming driver release.
Closing.

closed 2022-01-27 by bubnikv