prusa3d/PrusaSlicer #7278: Tool tip "Eject drive Ctrl+T" only appears after moving the mouse

fe60, 2021-11-09:
### Version
PrusaSlicer-2.4.0-beta1+win64-202110251333

### Operating system type + version
Windows 10 21H1 (19043.1288)

### Behavior
* Slice an arbitrary object
* export G-code to SD card
* notification about finished export appears
* move mouse pointer onto the `Eject` button (move it there and stop without further movement)
=> No tool tip
* move the mouse pointer a bit
=> Tool tip is shown

The tool tip should appear right away, when the mouse pointer hovers the `Eject` button (similar to the `Export to SD-card` button). 

_Is this a new feature request?_
No

lukasmatena, 2021-11-09:
@kocikdav

kocikdav, 2021-11-09:
Hello. Thank you for reporting. You are right, this should be fixed. But we are hopefully close to the release and bug introduced in fixing of this issue could significantly change GPU usage. So it is likely that fix will come after the release.

fe60, 2024-04-16:
Not yet fixed (2.7.4), but a minor issue and not a problem for me. Doesn't make sense to keep this open.

closed 2024-04-16 by fe60