prusa3d/PrusaSlicer #7280: can anyone add the hornet config and presets?

Medduus, 2021-11-09:
### Version
_2.3.3_

### Win10 + latest

### Artillery / Hornet + Marlin 2.0.7.2 (if known)

_Is this a new feature request?_ Yes

can anyone add all the hornet's configurations and profiles?

rtyr, 2021-11-11:
@SzabolcsHornyak is author of the Artillery bundle. Maybe he could add the profile for Hornet.

SzabolcsHornyak, 2021-11-11:
Hello Guys,
I don't have Hornet printer but I will try to request from the manufacturer. If they provide a printer I will do it ;)

Medduus, 2021-11-11:
I have some old cura profiles, that comes with the pendrive, also the firmware if needed

marine1988, 2022-01-03:
im waiting for this please any news?

Medduus, 2022-01-03:
i can provide the firmware and profiles used in ultimaker, don´t know how much that helps

bubnikv, 2022-01-04:
@Medduus we really need a printer to test on.

rtyr, 2022-01-14:
Artillery Hornet profiles will be included in the upcoming release. Closing.

closed 2022-01-14 by rtyr