prusa3d/PrusaSlicer #7282: PrusaSlicer-2.4.0-beta1 crashes after telling it to split object to parts

Mng-Michael, 2021-11-10:
PrusaSlicer-2.4.0-beta1+win64

Windows 10 Home 21H1

Prusa MK3S with MMU2S upgrade stock slicer settings

prusa slicer just crashes after telling it to split the stl to parts
load 3mf and split to parts
It should split the stl to parts but it just crashes after loading for 1-2 seconds
[image]
[prusaslicer 2.4.0 beta 1 crash.zip](https://github.com/prusa3d/PrusaSlicer/files/7508971/prusaslicer.2.4.0.beta.1.crash.zip)

FidelCapo, 2021-11-10:
I can confirm it's crashing.

lukasmatena, 2021-11-12:
This should be fixed in PrusaSlicer 2.4.0-beta2 (will be released shortly). Closing.

closed 2021-11-12 by lukasmatena