prusa3d/PrusaSlicer #7283: Feature request: Place seam in center of bed for better timelapse

pracovnik, 2021-11-10:
When using Octoprint's Octolapse, I am asked, where I would like the switch the layer (rebuild of gcode). And there is an option to set it in the center of the bed. 
As I can manipulate the object in PrusaSlicer, by moving the object on the bed. I can move it to center of bed whereas down-top is inside of the object. Additionally a nice option would be also to set the seam from vertical side, in case the object cannot be moved due to size or complexity.

_Is this a new feature request?_
YES

(1)
I would like to ask, to be able to place the seam start/end point to nearest as possible to center of the bed in PrusaSlicer.

Option like this:
[image]

(2)
I would like to ask to be able to place vertical "nearest as possible" point of seam start/end point on object.

[image]

bubnikv, 2021-12-10:
I am sorry, I don't understand. You may try in Czech.

pracovnik, 2021-12-17:
> I am sorry, I don't understand. You may try in Czech.

Vacsina rieseni pre robenia fotiek/videi pri tlaci(tisku) je riesena cez zmenu vrstvy. A chcel som poziadat, ci sa da dat, aby by som ukazal bod (graficky by som umiestnil vertikalny valec), kde by sa pri vypocte drahy snazil prusa slicer menit vrstvu. Pravdaze by slo aj o retrakciu v pripade ak by bol dany bod mimo objektu. Tym by sa dosiahol lepsi efekt na časozberne fotky pre video. 

Bod 1 by bola volba na stred podlozky, kde uz by bolo na mne poposuvat objekt, aby mal co najviac vertikalne plochy v nej, v strede podlozky. A tym sa dosiahne relativne stabilne casozberne fotky.

Bod 2, je podobne ako bod 1 ale by som povedal, kde ma byt priblizne menenie vrstvy a prusa slicer by podla toho prerataval cestu trysky.

lukasmatena, 2021-12-17:
Can't you just add an extra travel into Before layer change G-Code? I don't like the idea of another seam position just for better timelapses to be honest.

bubnikv, 2022-06-29:
We will not implement the request. Closing.

closed 2022-06-29 by bubnikv