prusa3d/PrusaSlicer #7284: Objects being removed from build plate after slicing.

jhenley01, 2021-11-11:
### Version
2.3.3

### Operating system type + version
MacOS Big Sur 11.2.1

### 3D printer brand / version + firmware version (if known)
Prusa Mini

### Behavior

I apologize if this has been brought up before but I was unable to find any previous reports of this issue.

If you have a lot of small parts on the build plate sometimes the 3D editor will show that all parts will fit on the build plate but after slicing some of the parts will be removed. This often happens when you use the option "Fill bed with instances"

I've attached screenshots of an example where I used fill bed with instances and then after slicing you can see that some of the outer parts got removed.

[image]
[image]

rtyr, 2021-11-11:
Could you please provide your 3MF project file? (File - Save Project as..., then archive it to the zip file and attach here)

tamasmeszaros, 2021-11-12:
duplicate of #5836

closed 2021-11-12 by tamasmeszaros

rtyr, 2021-11-12:
This issue is already fixed on master. The fix will be included in the upcoming 2.4.0-beta2.