prusa3d/PrusaSlicer #7285: The options spiral_vase and complete_objects are mutually exclusive, but PrusaSlicer doesn't warn or disable either when enabling the other feature

plampix, 2021-11-11:
### Version
2.4.0-beta1

### Operating system type + version
Debian GNU/Linux 11

### 3D printer brand / version + firmware version (if known)
Not relevant, but Prusa MK3S 3.10

### Behavior
When I've set complete_objects in my printing profile, and enable spiral_vase, PrusaSlicer tells me about infill, top layers etc. But it doesn't mention complete_objects, which is a conflicting option.

The other way around, if I've set spiral_vase in my printing profile, then enable complete_objects, it doesn't warn me spiral_vase is ignored.

rtyr, 2021-11-11:
> When I've set complete_objects in my printing profile, and enable spiral_vase, PrusaSlicer tells me about infill, top layers etc. But it doesn't mention complete_objects, which is a conflicting option.

It isn't conflicting option. You can use vase mode with complete individual objects enabled. It is actually the only way how to print more objects using vase mode.

> The other way around, if I've set spiral_vase in my printing profile, then enable complete_objects, it doesn't warn me spiral_vase is ignored.

It is not ignored. On the contrary, PrusaSlicer shows a notification in case you have more objects on the plater.

[image]

I believe you issues with the spiral vase/complete individual objects combination is probably about extruder clearance values. For example, if you want to print two 60mm cylinders sequentially in the vase mode, your extruder clearance height must be at least 60mm.

plampix, 2021-11-11:
Check out this cylinder, complete_objects is enabled. There's no G1 anywhere that does anything with Z:
[image]

And compare it to this one, with complete_objects disabled.
[image]

rtyr, 2021-11-11:
Thanks for clarification, we will check.

bubnikv, 2021-11-29:
Spiral vase worked in PrusaSlicer 2.3.3 in sequential mode, it was re-enabled in PrusaSlicer 2.4.0-beta2. Closing.

closed 2021-11-29 by bubnikv