prusa3d/PrusaSlicer #7286: Slicing vase-like objects can generate unsupported solid infill

cooperised, 2021-11-11:
### Version
2.3.3+linux-x64 (also tested on 2.4.0-beta1)

### Operating system type + version
Linux / Ubuntu 20.04.3 LTS

### 3D printer brand / version + firmware version (if known)
Prusa i3 MK3S

### Behavior
* When using PrusaSlicer to create hollow vase-like objects from solid STLs, if the base is not flat (and hence vase mode is not available), areas of unsupported solid infill can be generated.  To reproduce the problem, create a solid body whose base has one or more concave regions, then slice with 0% infill and no top layers.
* _Expected Results_
The solid infill used to thicken the base should be placed only in areas where there is material underneath it.
* _Actual Results_
Solid infill is placed in midair.  Cropped screenshot showing the layer below the rogue infill, with bridging correctly applied:
[image]
Cropped screenshot showing the very next layer, with solid infill clearly extending way beyond the supported area:
[image]

#### Project File (.3MF) where problem occurs
[infill_bug.zip](https://github.com/prusa3d/PrusaSlicer/files/7523198/infill_bug.zip)

bubnikv, 2021-12-10:
[image]

cooperised, 2026-09-18:
I get that there's been a rewrite but closing all open issues does seem like a strange way to address them. I will do some testing when I get time with 3.0.0a and close this manually if the issue is resolved.