prusa3d/PrusaSlicer #7287: UI not showing intended geometry after cut. if sliced part seems fine.

michael725stone, 2021-11-12:
PrusaSlicer
Version:   2.4.0-alpha2+win64
Build:     PrusaSlicer-2.4.0-alpha2+win64-202109241329
Surface Pro 7
Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 19042), 64-bit edition
Total RAM size [MB]: 16,746MB
OpenGL installation
GL version:   4.6.0 - Build 27.20.100.9621
Vendor:       Intel
Renderer:     Intel(R) Iris(R) Plus Graphics
GLSL version: 4.60 - Build 27.20.100.9621

Prusa MK3

### Behavior
Slicer not showing geometry as intended after cut

basically I import  parts and like to cut them to check fitment before commit to long print. I attempted to cut 1mm from the bottom of this part for slot and overall size verification. **when i cut the part the slots seem to have turned into protruding tabs when I intended to see negative space in the slots.** delete model and import over and over and cut shows incorrect. restart slicer and same effect. attempted to slice the file and it was CORRECT! (see pictures). This was quite annoying plain and simply because it was a visual problem that made me not want to continue as it clearly looked wrong. I hope the pictures will show you more.

As im writing this i found my slicer 2.3.0 does not have this issue while modifying the file.
wont get into crashes when slicing large files(has not been consistent)

Oddly annoying. thanks for reading.

[image]
[image]
[image]

bubnikv, 2021-12-08:
I believe this issue was fixed long time ago.

closed 2021-12-08 by bubnikv

michael725stone, 2021-12-08:
lol only took a month and i dont believe you fixed anything.

bubnikv, 2021-12-08:
lol thank you for your response.
No 3MF provided, we believe the issue was fixed.
Closing.