prusa3d/PrusaSlicer #7289: 2.4.0-beta1: Slice/Export Button does not look like a button, bring back the old design

fe60, 2021-11-12:
### Version
Version: 2.4.0-beta1+win64
Build: PrusaSlicer-2.4.0-beta1+win64-202110251333

### Operating system type + version
Windows 10 (build 19043), 64-bit edition

### Behavior
The changed design of the Slice/Export doesn't "feel" like a button to me. Without the padding and the border surrounding the grey area it doesn't look like a button. The changed highlighting (text only instead of the whole button) makes it even worse.

I don't know which this had been changed, but **I would like to get back the old design**.

Left - New 2.4
Right - Old 2.3 (the one I would like to get back)

[image]

**@DevTeam**
Feel free to close this ticket. It's about my personal preference. Or wait for some feedback from other users.

**@Users**
Vote using :+1: and :-1: .
Comments can't be aggregated and without additional explanation they only make it harder to read a discussion.

YuSanka, 2021-11-16:
It was made during implementation of the Dark Mode for Windows.
Thanks for your suggestion :) But we decided not to revert these changes. Instead of 1000 words, just take a look and see how it looks on Windows 11
[image]

fe60, 2021-11-19:
Looks good indeed on Windows 11. Unfortunately my PC does not meet the requirements to run Windows 11.

Hopefully the differences between Windows 10 and 11 remain such minor UI details.

closed 2021-11-19 by fe60