prusa3d/PrusaSlicer #7291: test_3mf fails on Archlinux

Salamandar, 2021-11-12:
### Version

2.4.0.beta1.r149.ge70fc7f0d

### Operating system type + version

Manjaro (Arch based), AUR package (i'm the packager)

### Behavior
During build, ctest -V fails, here is the relevant scenario output in the log file:
```
Testing Scenario: 2D convex hull of sinking object
    Given: model
     When: model is rotated, scaled and set as sinking
     Then: 2D convex hull should match with reference

~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
libslic3r_tests is a Catch v2.13.6 host application.
Run with -? for options

-------------------------------------------------------------------------------
Scenario: 2D convex hull of sinking object
      Given: model
       When: model is rotated, scaled and set as sinking
       Then: 2D convex hull should match with reference
-------------------------------------------------------------------------------
./aur/prusa-slicer-git/src/PrusaSlicer/tests/libslic3r/test_3mf.cpp:117
...............................................................................

./aur/prusa-slicer-git/src/PrusaSlicer/tests/libslic3r/test_3mf.cpp:118: FAILED:
  REQUIRE( res )
with expansion:
  false

    Given: model
     When: model is rotated, scaled and set as sinking
```

I tried to find the reason of the failure, but couldn't. So I'm pinging here to get help and maybe there's an issue in the code that only appears on some setups ?

Here's the cmake configuration command line :
```
    WX_CONFIG=wx-config-gtk3 \
    cmake .. -G Ninja \
        -DCMAKE_INSTALL_PREFIX=/usr \
        -DCMAKE_INSTALL_LIBDIR=lib \
        -DSLIC3R_FHS=ON \
        -DSLIC3R_PCH=ON \
        -DSLIC3R_WX_STABLE=ON \
        -DSLIC3R_GTK=3 \
        -DSLIC3R_STATIC=OFF \
        -DOPENVDB_FIND_MODULE_PATH=/usr/lib/cmake/OpenVDB

```

bubnikv, 2021-11-12:
@enricoturri1966 ?

enricoturri1966, 2021-11-15:
Unfortunately I do not have the hardware to try to reproduce the issue.
Looking at the code it may be a float precision issue.

Could you please try to modify test_3mf.cpp by adding these lines:

```
            for (const Point& p : hull_2d.points) {
                std::cout << p.x() << ", " << p.y() << "\n";
            }
```

before

`            bool res = hull_2d.points == result;`

and check if the output on console differs from:

Testing Scenario: 2D convex hull of sinking object
    Given: model
     When: model is rotated, scaled and set as sinking
-91501496, -15914144
91501496, -15914144
91501496, 4243
78229680, 4246883
56898100, 4246883
-85501496, 4242641
-91501496, 4243
     Then: 2D convex hull should match with reference

bubnikv, 2021-12-08:
Fixed with d6bb8eead91a4b852e362e2cb4d67f6c3402fc87

closed 2021-12-08 by bubnikv