prusa3d/PrusaSlicer #7292: Feature Request : Option to set X;Y offset distances for "add instance"

revaip, 2021-11-12:
### Version
all

### Operating system type + version
not related

### 3D printer brand / version + firmware version (if known)
not related

### Behavior
* _Expected Results_
The offset of the added instance should be set manually.

* _Actual Results_
In PrusaSlicer the "add instance" function adds the new instance offseted from the most recent instance. This is correct, but the offset is fixed, cannot be set, furthermore, it applied both X and Y coordinates. The offset is different for all printer sizes, so not so easy to calculate and re-position the new instances manually... especially when the offset is fraction (SL1:5,9mm). In most cases the fill bed is not an option, because we need to put the parts under each other, or [with overlapping pads](https://user-images.githubusercontent.com/90909096/141468759-d9d529d7-d34b-4f6b-962f-098c31099ad0.png), or [inside of each other](https://user-images.githubusercontent.com/90909096/141467825-2d185068-76a3-4387-9999-6e9f4ae38992.png)...

The offset seems to be related to the bed size:

#### For FFF printers
MK3S (bed size: 250x210mm): the offset is 12,5mm
Mini (bed size: 180x180mm): 9mm
custom printer with 100x200mm bed: 10mm
custom printer with 200x100mm bed: 10mm
custom printer with 100x100mm bed: 5mm
custom printer with 1x500mm bed: slicer freezes
custom printer with 10x300mm bed: 15mm
custom printer with Ø200mm bed: 10mm

The add instance offset is 5% of the X or Y dimension of the bed, which is longer (eg. 100x200mm: 200*0.05=10mm).

#### For SLA printers:
SL1 (printable area size: 120.96x68.04mm): 5,9mm
The platform 3d model (sl1_bed.stl) is 130x72mm, which gives 6,5mm at 5%, which is not equal to the 5,9mm offset.

SL1S (printable area size: 128x81mm): 6,35mm, which is 4,96%...
The platform 3d model (sl1s_bed.stl) is 129x82mm, which gives 6,45mm at 5%, which is still not equal.

I cannot figure out the relation between SLA printer bed/platform/printable area size and offset.

[image]

Thank you very much!

_Is this a new feature request?_
yes

pintail120, 2021-11-23:
is this feature request for control over where the "add instance" delivers the next part in the X / Y
currently I assume there is no access to set where the next instance is placed ?

lukasmatena, 2021-11-24:
I personally don't think we need two extra configuration options for something like this. The offset will still be applicable only to objects with a specific shape and size (you yourself provided examples proving that the values need to be different for almost every distinct object). The 'Add instance' option was meant for adding instances, while sensible arrangement should be taken care of by arrange (although you are right that it currently does not place objects into holes inside other objects etc).

As for the the value of the offset, it is now calculated as 5 % of bed width or height, whichever is greater. That gives 0.05*118 mm for SL1, 0.05*127 mm for SL1S). The main reason the offset is there is that the instance is not added over the existing one, so the user can see that it was indeed added.

dartrax, 2022-06-30:
This should be closed as duplicate of #5890.

closed 2022-07-17 by stepikovo