prusa3d/PrusaSlicer #7293: Do not put __FILE__ info in release mode

Salamandar, 2021-11-12:
I couldn't reopen https://github.com/prusa3d/PrusaSlicer/issues/2243 so i'm creating a new issue.
It's once again relevant : https://github.com/prusa3d/PrusaSlicer/blob/master/src/slic3r/GUI/3DScene.hpp#L16

bubnikv, 2021-11-12:
I don't understand. HAS_GLSAFE is only enabled in debug mode, not in
release mode.
What exactly is the problem this time?

Salamandar, 2021-11-13:
@bubnikv this was fixed just before you commented by @enricoturri1966 .

lukasmatena, 2021-11-15:
@Salamandar Well, there is another use of the `__FILE__` macro in [SendSystemInfoDialog.cpp#L567](https://github.com/prusa3d/PrusaSlicer/blob/5dfba04e0238ca4a4e84b3f46d57d77bcd386578/src/slic3r/GUI/SendSystemInfoDialog.cpp#L567). And to be honest, I don't understand the problem. The macro is available in Release mode and "it may be a good idea to use only in Debug" is not especially convincing. Is there some actual reason to not use it? I mean, I can of course replace it with the filename (which will break after the file is renamed, but it is quite unlikely to happen in this case), but I'm curious. Thanks.

Salamandar, 2021-11-16:
@lukasmatena Thanks, didn't see that.

Well, the rationale is that when packaging a software, you don't want the final binaries to contain the paths of the source code on the build machine. That also breaks the "reproducible builds" paradigm that more and more distros try to achieve.

That path is only relevant to developers. Even most people that built the software themselves may have deleted the source and will probably never read it.
About SendSystemInfoDialog.ccp, you finally just use the filename itself without the path. You could either use __BASE_FILE__ or use the gcc preprocessor option -fmacro-prefix-map=OLD=NEW (see https://blog.conan.io/2019/09/02/Deterministic-builds-with-C-C++.html)

lukasmatena, 2021-11-16:
@Salamandar Thanks. I think I'll just do this: 0971b27, that should make everyone happy. I'll commit this into master after I actually test with gcc. I really don't want to solve something this minor with platform-dependent code.

lukasmatena, 2021-11-16:
@Salamandar After 2c31755, the macro should only be evaluated in Debug mode. I admit that it is even shorter and easier to read. Closing.

closed 2021-11-16 by lukasmatena

Salamandar, 2021-11-16:
Thank for the fix !!