prusa3d/PrusaSlicer #7296: Random crash to desktop Beta 2.4.0

Crowlord, 2021-11-12:
### Version
Beta 2.4.0 Win64

### Operating system type + version
Windows 10 PRO 21H1
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
N/A

### Behaviour
Random crash to desktop, no error message. Seems to happen when asked to process. 

#### Project File (.3MF) where problem occurs
N/A (no set behaviour)

fe60, 2021-11-15:
Its hard to analyze/fix an issue that can't be reproduced. Did you try running PrusaSlicer with logging to get more information?
See comment from developer in https://github.com/prusa3d/PrusaSlicer/issues/4316#issuecomment-634769532

Crowlord, 2021-11-15:
Running it that way now, lets see if i can get some info. :)

Crowlord, 2021-11-15:
just had my first one. There is no log error at all it just shuts!

+++++++++++++++++++++++++++

* Mark bundle as not supporting multiuse
< HTTP/1.1 100 (Continue)
* We are completely uploaded and fine
* Mark bundle as not supporting multiuse
< HTTP/1.1 201 CREATED
< Content-Type: application/json
< Content-Length: 312
< Location: http://ian.local/api/files/local/Worm_B_0.2mm_PLA_MINI_5h18m.gcode
< Cache-Control: no-store, no-cache, must-revalidate, post-check=0, pre-check=0, max-age=0
< Pragma: no-cache
< Expires: -1
< X-Clacks-Overhead: GNU Terry Pratchett
< X-Robots-Tag: noindex, nofollow, noimageindex
< X-Content-Type-Options: nosniff
< X-Frame-Options: sameorigin
<
* Connection #0 to host ian.local left intact
[2021-11-15 07:49:45.793660] [0x00005d5c] [debug]   OctoPrint: File uploaded: HTTP 201: {"done":true,"files":{"local":{"name":"Worm_B_0.2mm_PLA_MINI_5h18m.gcode","origin":"local","path":"Worm_B_0.2mm_PLA_MINI_5h18m.gcode","refs":{"download":"http://ian.local/downloads/files/local/Worm_B_0.2mm_PLA_MINI_5h18m.gcode","resource":"http://ian.local/api/files/local/Worm_B_0.2mm_PLA_MINI_5h18m.gcode"}}}}

PS D:\OneDrive\Desktop\PrusaSlicer-2.4.0-beta1+win64-202110251333>

lukasmatena, 2021-11-15:
@Crowlord Thanks, but we need more information. Most importantly we need to know what exactly you did before the crash. "Random crashes" is by far not enough.

"Seems to happen when asked to process." - ok, so then it is not random, but it happens after you did... something. We need to know what. The log indicates you were sending something to OctoPrint? Can this be a duplicate of #7170?

Crowlord, 2021-11-15:
OK.

It has happened when I have changed print profiles, when  I have finished
exporting, mid exporting, when the program is lieing unused. It has been
random with no obvious cause.

I have experienced this problem for some time before raising it, in fact I
came away from the alpha for this very reason assuming somebody else would
pick up on it. As its now beta and could make it into production I thought
i could possibly start to raise a concern.

If I had no detailed and specific info I would have shared it.

My own personal testing makes me suspect it is One Drive causing an error.
I have moved the beta to an un-backed up location on my SSD and will report
back.

To be honest I half expected snark BECAUSE I have so little info but was no
longer prepared to leave a potentially breaking bug unreported. To have got
some is... frustrating.

closed 2021-11-15 by Crowlord

lukasmatena, 2021-11-15:
@Crowlord If it happens even when you are not uploading to OctoPrint and if it happened in alpha1, than it is not just #7170, but something else. #7170 was a bug introduced in beta1 only. You can start by running slicer from command line as `prusa-slicer-console.exe --sw-renderer`. In case that fixes the crashes, the problem is somehow connected to the graphics driver.