prusa3d/PrusaSlicer #7299: Something about how this part is color-painted makes PS-beta1 crash on slice

SomeoneElse58, 2021-11-13:
### Version
2.4.0-beta1

### Operating system type + version
Win10 Pro 21H1

### 3D printer brand / version + firmware version (if known)
MK3S MMU2S with TZB 3.3.0 firmware

### Behavior
Painted a small part of the [Puddle Jumper](https://www.prusaprinters.org/prints/86439-puddle-jumper-from-stargate), as shown below, and PrusaSlicer will crash immediately after clicking the slice button.
[image]

Changing the color for the bottom still results in a crash, so it's not that.
[image]

Removing all painted areas will result in a successful slice.

#### Project File (.3MF) where problem occurs
[2.4.0b1_crashesOnSlice.zip](https://github.com/prusa3d/PrusaSlicer/files/7531653/2.4.0b1_crashesOnSlice.zip)

lukasmatena, 2021-11-15:
Reproduced, thanks. We will fix it.

SomeoneElse58, 2021-11-16:
Looks like rotating the part to z=90 will still crash on slice. However, rotating to z=270 or z=180 will successfully slice.

So yea, weird bug.

hejllukas, 2021-11-17:
Thank you for reporting this issue. It was fixed with 9cac904f971d28599b5cf01356f91164b88b34d8, and it will be part of the upcoming beta release.

closed 2021-11-17 by hejllukas