prusa3d/PrusaSlicer #7301: Endless loop  telling me 'Supports work better if the following feature is enabled: bridging perimeters' when setting 'generate support material' on a single object

plampix, 2021-11-13:
### Version
2.4.0-beta1

### Operating system type + version
Debian GNU/Linux 11

### 3D printer brand / version + firmware version (if known)
Not applicable

### Behavior
Make sure support is off, detect bridging perimeters is off as well.
Add a Cylinder
Using the context menu, add 'Support material' to this object, and check 'Generate support material'.
A popup will appear
[image]
Unfortunately there's no option to enable this option in this popup, like there is when you enable support from the print settings:
[image]

The latter dialog also allows you to say 'No', and the slicer will continue. On the first dialog, if you click ok, the dialog closes, and immediately reopens. This will require a forced closing of the program, and you'll lose everything you've done until then.

n8bot, 2021-11-14:
I experienced the same thing, but I cannot remember the circumstances.

Are these intrusive messages really necessary? There must be a better way to communicate to the types of users that are being targeted with these messages.

dartrax, 2021-11-14:
I can't reproduce on Win 10 x64, may be OS specific.
_May_ be related to #6992.

n8bot, 2021-11-14:
> I can't reproduce on Win 10 x64, may be OS specific. _May_ be related to #6992.

I experienced this issue on Win 10 64.

dartrax, 2021-11-14:
Could depend on other settings (GUI preference etc.) like in #6992. I tried with different settings but I may have missed some.

plampix, 2021-11-14:
Yes, sorry, in one of my printer profiles I've got 'Detect bridging perimeters' disabled. Only then do you get the dialog when you enable support.

dartrax, 2021-11-15:
I know, I also get the dialog as you described, but it doesn‘t come again after clicking OK. What‘s your GUI setting? Do you have background processing enabled?

YuSanka, 2021-11-15:
Thanks for your report. This bug is fixed and will be a part of upcoming release.

closed 2021-11-15 by YuSanka

plampix, 2021-11-15:
I can confirm this fixes the problem. It would be 'nice to have' the option to add an object specific 'Detect bridging perimeters' setting to the object in question.

n8bot, 2021-12-06:
I have encountered this endless loop once again.

The "Support works better, if..." dialog enters an endless loop...

If you are in a scenario where the dialog has asked you once already if you want to detect bridges, and you select no... then you add a modifier to a model. Then, you attempt to add the setting Fuzzy Skin, it will bring up the endless loop dialog -- only OK button, which dismisses it then it pops back up. 

[image]

It enters the endless loop even if you disable the else statement:

https://github.com/prusa3d/PrusaSlicer/blob/bfce4f69019d2ddd2ffd4dbbacc8ef5d4f32f015/src/slic3r/GUI/ConfigManipulation.cpp#L183-L186

Which needs to be disabled, because otherwise the popup comes every single time supports are enabled while detect bridging perimeters is not.