prusa3d/PrusaSlicer #7311: Keyboard shortcut for "Reload from disk" (F5) not working on Linux/Ubuntu

kankaristo, 2021-11-16:
### Version
2.3.1+linux-x64

_Use `About->About PrusaSlicer` for release versions_   <-- This is `Help -> About PrusaSlicer`, by the way

### Operating system type + version
Pop!_OS 21.04 (based on Ubuntu)

### 3D printer brand / version + firmware version (if known)
Prusa Mini, firmware 4.3.1+2759 (most likely not relevant for the bug)

### Behavior
Reloading the STL file with `Edit -> Reload from disk` works fine, but the keyboard shortcut to do the same (`F5`) does nothing.

#### Project File (.3MF) where problem occurs
(project doesn't matter, happens for all projects)

plampix, 2021-11-18:
This does work in a build from master, so this will work when 2.4 is released.

plampix, 2021-11-18:
Well, my last comment wasn't entirely true. If I load an object, F5 won't reload it from disk. If I reload it using the Edit menu, it reloads, and after that initial reload, F5 will then work as expected.

Even after deleting the object, adding a new one, F5 won't initially work, only after a reload from the menu.

fe60, 2021-11-26:
I can't reproduce the issue with a VM and a fresh installation of [Pop!_OS](https://pop.system76.com/) 21.04.
Both app images provided for PrusaSlicer [2.3.3](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.3.3) work as expected. `F5` triggers reload right from the beginning.

That's what I did:
1. Import object using Ctrl+i and keep the object selected
2. Replace imported object in file system
3. switch back to PrusaSlicer and press `F5`
=> reloads as expected

kankaristo, 2021-11-26:
@fe60, sorry, there seems to be one more thing that's required to reproduce this: you need to be in "Preview" mode. At least there, `F5` does nothing, but `Edit` > `Reload from disk` works as expected (although you need to click "Slice now" again to see the reloaded objects). This might not even be Linux specific.

fe60, 2021-11-26:
`F5` doesn't work at all in Preview on Windows, while `Edit > Reload from disk` does (tested with 2.3.3 and 2.4.0-beta1).
The reload option offered in the object specific settings menu works in the Preview too.
[image]
Screenshot shows 2.3.3

Personally I don't expect it to work in the Preview, but the behaviour seems to be inconsistent.

kankaristo, 2021-11-26:
I would expect it to work unless the option under the `Edit` menu is grayed out. And I would expect it to work in  Preview mode as well (both through the menu and with `F5`), although it would be nice if it would automatically re-slice as well, since currently Preview mode just goes empty on reload.

I think it would be confusing if reload wasn't available at all in Preview mode, especially for a new user, since there wouldn't really be any good way of telling the user why the option is unavailable.

neophyl, 2021-11-26:
One problem is that you can’t select an object on the plater in preview mode, as it’s a preview of the output gcode, not the objects.  That why the menu works, it’s operating on the object previous selected and also why the function using the object tree is available. You have something to select there to define exactly “what” is reloaded. You don’t with the plater. 

If you don’t want the plater to be empty you need to enable the automatic slicing. Personally I hate the auto slice so have it off, I just use the reslice keyboard shortcut to force a reslice when I’m on the preview

yahbluez, 2023-09-12:
This issue still occurs in 2.6.1 on manjaro linux.

closed 2026-09-29 by prusaslicer-bot[bot]