prusa3d/PrusaSlicer #7312: Direction of solid infill

MKaiser96, 2021-11-17:
Hello,

I wish there will be an option to define the direction order of solid infill.
For example: [0 deg, 90 deg, 45 deg, -45 deg]

This would be useful for isotropic material behavior.

SomeoneElse58, 2021-11-18:
Is this not already a thing?
[image]

neophyl, 2021-11-18:
Not quite no.  While that setting allows you to set the initial angle for the infill, each successive layer is placed at 90 degrees (well for ones like rectilinear or monotonic etc) so that they cross hatch.
The original post is not particularly specific but I think they want control over the angle of each layer as its put down.  That way you can make sure all the layers are printed in the same direction.  For example setting  0,180 degrees would align all the lines the same way as the first layer would be at 0 degrees, the second at 180, the third at 0 again.
While this isn't best for strength it is often used when printing with 'clear' or translucent filaments to give maximum light transmission.

I asked for basically the same feature in Super Slicer for those same reasons.  You can also use it to do some nice artistic things with infill patterns (zero perimeters) by setting an initial angle and then a smaller offset like 45,10, so that it uses 45/65/75/85 to give nice spiral effects with infill patterns like hexagonal.

lukasmatena, 2021-11-18:
@MKaiser96  Can you please tell us more details about your use case? Is it the one that @neophyl suggested?

n8bot, 2021-11-18:
I think the user simply wants control of the solid infill angle.

It's possible they also want granular control of the layer angles for all infill. But they specifically state solid infill.

MKaiser96, 2021-11-19:
I think I did not express myself correctly. I want to set the order the order of the fixed filling. For example like this:

1. 0 deg
2. 90 deg
3. 45 deg
4. -45 deg
5. 0 deg
6. 90 deg
7. 45 deg
8. -45 deg
9. ...

In Cura this function is called _top/bottom line directions_

rtyr, 2021-11-19:
Thanks for clarification. There is already an issue requesting this, so I will close this as a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/2293.

closed 2021-11-19 by rtyr