prusa3d/PrusaSlicer #7313: Crash while sliceing a SLA Print

cybergurke, 2021-11-17:
### Version
working: 
PrusaSlicer-2.3.3+linux-x64-GTK3-202107161044.AppImage 
PrusaSlicer-2.4.0-alpha1+linux-x64-202109011425.AppImage
crashing:
PrusaSlicer-2.4.0-alpha2+linux-x64-202109241328.AppImage
PrusaSlicer-2.4.0-alpha3+linux-x64-202109302024.AppImage
PrusaSlicer-2.4.0-beta1+linux-x64-202110251331.AppImage
compiled from master version_2.4.0-beta1-185-g8a4323c5b

### Operating system type + version
OS: Arch Linux x86_64
Kernel: 5.15.2-zen1-1-zen 
DE: Plasma 5.23.3
CPU: AMD Ryzen 5 1600 (12) @ 3.193GHz
GPU: NVIDIA GeForce RTX 2070 SUPER

### 3D printer brand / version + firmware version (if known)
Prusa SL1S

### Behavior
* _Describe the problem_
PrusaSlicer crashes while slicing a SLA print since alpha2 with this Error:
Im tested with a new profile.
```
[2021-11-17 20:41:05.225730] [0x00007f2141f53780] [error]   Uncaught exception: basic_string::substr: __pos (which is 35) > this->size() (which is 27)
Unhandled unknown exception; terminating the application.
[2021-11-17 20:41:05.312643] [0x00007f2141f53780] [error]   Uncaught exception: basic_string::substr: __pos (which is 35) > this->size() (which is 27)
[2021-11-17 20:41:05.317937] [0x00007f2141f53780] [error]   Uncaught exception: basic_string::substr: __pos (which is 35) > this->size() (which is 27)
Unhandled unknown exception; terminating the application.
fish: Job 1, './PrusaSlicer-2.4.0-beta1+linux…' terminated by signal SIGABRT (Abbruch)

```

* _Steps needed to reproduce the problem_*
- select a SLA printer
- add cube
- press slicing 
crash

lukasmatena, 2021-11-18:
Thanks, reproduced. We will fix it.

kocikdav, 2021-11-18:
Thank you for reporting. Should be fixed now.

closed 2021-11-18 by kocikdav

cybergurke, 2021-11-18:
Thank you. It works.