prusa3d/PrusaSlicer #7314: Feature Request: Import .Mat texture alongside .Obj file for Multi-Part Painting

wenbilliams, 2021-11-18:
_Is this a new feature request?_
Yes

It would be incredible to be able to load a model with an albedo texture into PrusaSlicer and have the slicer automagically paint each face of the model accordingly.

This could cause issues with the conversion of full-scale colors to the finite (5) colors present in the MMU2S attachment. I propose a solution similar to how Canvas3d converts stamp images to the available colorset.

I'm happy to work on coding this feature if any experienced developers could point me at a few spots to look at. I have plenty of C++ experience. I'm currently trying to dig up the painting tool PR to get at the bits and pieces relevant to applying material to a surface.

Object/Material import reference (shapeways article): https://support.shapeways.com/hc/en-us/articles/360023735394-Full-color-3D-printing-with-3DS-Max
Image import / Color filtering reference: https://www.youtube.com/watch?v=rrb-Zn0iLdQ

wenbilliams, 2021-11-19:
@hejllukas I see you've done a bunch of PR around multi material painting. Can you point me at any spots in the code that could help me get started writing this?

hejllukas, 2021-11-23:
It sounds to me like a cool feature that could simplify printing multi-material objects. Sorry for the later reply.

The multi-material painting consists mainly of three separate parts:
- _slic3r/GUI/Gizmos/GLGizmoMmuSegmentation.{hpp, cpp}_ - This mainly provides the UI for painting and rendering all painted triangles through the method _TriangleSelectorMmGui::render()_.
- _libslic3r/TriangleSelector.{hpp, cpp}_ - Common to all painting gizmos (supports, seams, and multi-material). This provides a backend for triangle painting and triangle splitting (if it is enabled).
- _libslic3r/MultiMaterialSegmentation.{hpp, cpp}_ - This provides segmentation of a model according to the painted triangles. This part will probably not need to be changed for your purpose.

Each model is represented as a _ModelObject_ that contains at least one _ModelVolume_.
Each _ModelVolume_ keeps a _TriangleMesh_ with a representation of that model.
The _ModelVolume_ class contains the variable _FacetsAnnotation mmu_segmentation_facets_ with serialized painting and splitting of triangles from a _TriangleMesh_. The encoding of stored data is described more in _TriangleSelector::serialize()_. In short, each triangle is encoded using either 4 or 8 bits (This was chosen because of backward compatibility with support and seams, which used 4 bits and supports up to 4 different states/colors). These bits encode if a triangle is split or not, and it is color (stored as _EnforcerBlockerType_). 
Currently, the number of different colors is limited to 16 (default color/extruder and 16 other colors).

In _MultiMateraiSegmentation.cpp_, all _ModelVolume_ are iterated, and method _FacetsAnnotation::get_facets()_, is called, which according to the passed _EnforcerBlockerType_ will only return triangles of the selected color from the data stored in _mmu_segmentation_facets_.
To assign a color to a triangle, you can call _TriangleSelector::set_facet()_, or _TriangleSelector::Triangle::set_state()_. Both these methods must always be called on triangles that aren't split/divided.
The painted trinagles from the _TriangleSelector_ are then serialized into previously mentioned variable _ModelVolume::mmu_segmentation_facets_ by calling _FacetsAnnotation::set()_.

_EnforcerBlockerType_ can have values from 0 to 16. 0 is the default color of the model (default extruder). This means that the triangle isn't painted, and there is no need to assign this default color explicitly. Values from 1 to 16 are individual colors/extruders.

If I understand it correctly, .obj files use **usemtl** to assign the material to faces below this command, then at least in the beginning, there probably would be no need to parse the .mat/.mtl.
Also there is also no need to further split/subdivide the triangles into finer ones for this feature. So you wouldn't need to manipulate the triangle splitting/subdivision in _TriangleSelector_ in any way. Nevertheless, for completeness, I will describe it below.

I think you will need to look at _libslic3r/Format/Obj.{hpp, cpp}_ and _libslic3r/Format/objparser.{hpp, cpp}_ where the .obj files are parsed. I saw that there is some support for **usemtl**, but the read data isn't processed anymore.
So I guess you will need to extend _load_obj_ somehow and figure out how to store the materials.

I think the methods _TriangleSelector::seed_fill_select_triangles()_ and _TriangleSelector::seed_fill_apply_on_triangles()_ could be used as inspiration and to get familiar with _TriangleSelector_. _TriangleSelector::seed_fill_select_triangles()_ is called with starting facet, from its go through all adjacent facets (based on the angle between them) using a queue and flagged that were selected by seed fill (in UI is called **Smart fill**).
Then, in _TriangleSelector::seed_fill_apply_on_triangles()_, all go through all triangles and assign the selected color/_EnforcerBlockerType_ to all flagged triangles.

When _TriangleSelector_ is created, it copies vertices to _m_vertices_ and triangles to _m_triangles_. After initialization, all triangles are always unsplit/undivided, and _m_vertices_ and _m_triangles_ have the same data as they were in _TriangleMesh_.
New vertices and triangles are created only when a user paints with brushes with triangle splitting enabled (_TriangleSelector::select_patch()_).

**Triangle splitting**
I don't think this will be needed to implement this feature, but I will still quickly describe how the triangle splitting works.
When a user paints with brushes (in the code called _Cursors_), it is called the method _TriangleSelector::select_patch()_.
That assigns a color to a triangle if it is entirely inside a brush. If the triangle is only partially inside a brush, it can be split (unless a resulting edge will be smaller than a given limit determined from the brush diameter).
Splitting of a triangle is done through calling methods _TriangleSelector::split_triangle()_ and _TriangleSelector::perform_split()_.
Each triangle is always split into two, three, or four triangles. Triangles are split recursively, and their edges are always split in the middle. New vertices created by edge splitting are shared between adjacent triangles of the corresponding edge, so there should be no duplicate vertices in _m_vertices_.

One drawback to the above-described method of triangle splitting is that it results in painted areas on the brush border always being jagged. This causes issues on flat top and bottom layers, where larger holes in obj [... cut ...] der of two different colors. We invested some time to find a better solution, but unfortunately, the right solution to this problem isn't quite easy to do it right and stable.

I hope I've mentioned everything that might be useful about the multi-material painting/segmentation background and that it helps you.

I personally think that the most work will be with creating some kind of dialog or gizmo to assign colors/extruders to parsed materials and some user-friendly integration into the slicer.

wenbilliams, 2021-11-23:
@hejllukas Thanks so much for the in-depth description. This really helps me get started!

wenbilliams, 2021-11-27:
@hejllukas Where can I find sample code to read in .jpeg and .png files in this repo? I have parsed the mtl, and I'm trying to use Boost::GIL but struggling. If this repo has OpenCV I know how to use that to easily parse any image file.

wenbilliams, 2021-11-27:
And one last question - how do I publish a branch of this? Do I need to get a permission for this repo, or do I need to fork and then pull request from the other?

hejllukas, 2021-11-29:
> @hejllukas Where can I find sample code to read in .jpeg and .png files in this repo? I have parsed the mtl, and I'm trying to use Boost::GIL but struggling. If this repo has OpenCV I know how to use that to easily parse any image file.

We don't use OpenCV. We load .png using wxImage::LoadFile from the WxWidgets library. It should support .jpeg/jpg as well.
Parts of the GLTexture::load_from_png() function could serve as an example of how to load a .png file using wxImage::LoadFile and get its RGBA representation. But wxImage::LoadFile is also used in other places, or more information about wxImage::LoadFile is in the documentation for WxWidgets.

> And one last question - how do I publish a branch of this? Do I need to get a permission for this repo, or do I need to fork and then pull request from the other?

Yeah, you'll have to make a fork and then make a pull request from it (Described, for example, there: https://docs.github.com/en/pull-requests/collaborating-with-pull-requests/proposing-changes-to-your-work-with-pull-requests/creating-a-pull-request)

PaulKC, 2022-04-03:
That sounds really interesting. Have you found the time to implement something along these lines?

bubnikv, 2022-04-05:
@[wenbilliams](https://github.com/wenbilliams)
Multi-material segmentation is on our radar, however the issue we are facing is that we want to combine various sources of color segmentation (multiple volumes, modifiers, upcoming text) with the texturing, which is highly non trivial and we did not decide yet how to proceed.

TylonHH, 2022-08-19:
Hi
I have an complex model (architecture) which is painted and exported to 3MF or 3DS. It’s one STL with five colours.
How can I print this multimaterial? Painting again within prusa slicer is no option.
Could this FR help?
Cheers

JGarrett7, 2025-03-21:
@wenbilliams know texture assets better than I do, but assuming I am understanding what he is suggesting, I would like to +1 this idea. The ability to wrap a model with a texture consisting of 5 single colors in a program like Blender and import that into PrusaSlicer instead of filament painting would be an amazing feature. There are so many texture assets for 3D models that even limiting to those with no more than 5 colors would still open up a whole new world of options.

closed 2026-09-29 by prusaslicer-bot[bot]