prusa3d/PrusaSlicer #7315: Feature request: Option for automatic bridging for holes in the models

wasyl00, 2021-11-18:
Here is the idea. Many times when I model I have a problem that I'm forced to use support in case where there is a hole which is not on the bottom layer (for screw for example) in the model. Most of the time I go back and add one layer thickness cover to the bottom of the hole  just to make a perfect bridge and then I puncture the hole in the post. This is so much cleaner and faster solution then cleaning support material. This also saves me a lot of plastic in the process. I hope I explained clearly if not I could provide picture to show what I'm talking about. I wish there was an option in slicer to do it automatically.

DaTTcz, 2021-11-21:
It need an advanced setting about the size of the hole for which it will apply. Or better, add as another object modifier.

bobbyh41, 2021-11-21:
This is a great idea, I also do the same thing with a one layer sacrificial layer. Would be great if this was somehow an option in the slicer.

dartrax, 2021-11-21:
I also like that Idea, I've done that also in the past. But even better (however more complicated to implement) would be to have some kind of smart bridging that first prints tangent bridges in two directions around the hole before it prints the hole. I've "forced" it to do so by modeling those tangent bridges in CAD in the past:

[image]

[image]

Sandiegoo, 2021-12-04:
@wasyl00 I had the exact same idea and i wanted to write a feature request as well. This will be the feature that i would use the most if it will be implemented.

lukasmatena, 2021-12-06:
The use case quite resembles #1883.

I don't believe that covering holes as suggested in the first post is something that slicer should do on itself, not to mention that it might not be able to distinguish places where it is "desirable" and where it isn't.

ledvinap, 2021-12-06:
@lukasmatena : It may be possible to implement 'bridge extension' - generate bridge over unsupported areas (as is currently done), then extend it until it is fully supported. But polygon manipulation would be quite hard (IMO even splitting  support space for two bridges is quite hard and unimplemented).

BTW: object modifiers will probably take care of "desirable"

wasyl00, 2021-12-06:
Example of the thin layer I construct to bridge the gaps. This is so much more cleaner than having hole through material plus support.  This does not have to be limited to only circular shapes.  
[image]
[image]

bubnikv, 2021-12-08:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/1860

dartrax, 2026-09-18:
Still a valid FR