prusa3d/PrusaSlicer #7318: Object order is ignored (in non-CIO mode)

af12689, 2021-11-20:
### Version

Version 2.4.0-beta1+linux-x64

### Operating system type + version

OpenSuse Leap 15.1

### 3D printer brand / version + firmware version (if known)

Prusa i3 MK3S

### Behavior

The object order as defined by the object list is only honored if "Complete individual objects" (CIO) is enabled.

_Actual Results_

The attached screenshot shows the objects "0".."9" quasi randomly positioned on the build plate. With CIO enabled the object "0" is printed first, then "1", then "2", and so on, just like it is defined in the object list.

[image]

If CIO is disabled, then (per layer) the print order is a zig-zagging line starting at the bottom right ("6", "8", "5", "3", "2", "9", "7", "4", "0", "1"). The object order in the object list is ignored.

[image]

_Expected Results_

In non-CIO mode the object order should also be honored.

_Steps needed to reproduce the problem_

1. Disable CIO
   - Add more than one object to the plate
   - Arrange them in some way, order the objects in the list in some way
   - Slice -> Objects are printed in a zig-zagging line, the object order is ignored
2. Enable CIO
   - Slice -> Objects are printed in the order defined in the object list
3. Change order of objects
   - Slice -> Objects are printed according to new order
4. Disable CIO
   - Slice -> Objects are printed again in a zig-zagging line, the object order is ignored again

_Is this a new feature request?_

No

_Remarks_

This issue is similar to #5175, but #5175 is a feature request related to instance order, while this issue describes a problem with the existing object order functionality.

#### Project File (.3MF) where problem occurs

[af-ordertest.3mf.gz](https://github.com/prusa3d/PrusaSlicer/files/7573769/af-ordertest.3mf.gz)

viper0078, 2021-11-20:
Personally, I would prefer the current software specification.
If the order is to be maintained in non-CIO mode, it should be possible to change it with an option switch.

af12689, 2021-11-20:
That is a good idea.

For consistency this "use object print order" switch should be independent from the CIO switch.
This way the user can always decide if they want to force a specific print order or if they want the slicer to set the print order. The user does not need to fiddle with the ordering in CIO mode if the default ordering is good enough. The zig-zagging ordering that I have seen in non-CIO mode would be good for CIO mode.

rtyr, 2021-11-21:
Let's say I want to print objects on the screenshot. I added them randomly, because I just want to print them. With the behavior requested in this issue I would have to manually reorder objects in the object list for almost every multi-object print, it is not acceptable.

It would indeed have to be optional feature, something like "Force object print order". I will rename this issue and mark it as FR.

[image]

Tupson444, 2023-04-27:
First of all, of course this should be optional, as in most cases it is better as it is.

But we should definitely have this option, in the case we want it to print in certain order, for whatever reason. One example:

When multiple small items are printed (so small that they have no infill), perimeters are printed in certain direction (e.g. clockwise). At the end of printing one object, the nozzle is located say on the left of the object. So, then it moves to the next object which is to the right, and thus automatically "wipe" the nozzle into the part by crossing. However, in the next row of parts, it moves from the right to the left and thus the nozzle goes straight from the external perimeter of previous object and ooze that way. The result is that rows of objects that are printed left to right look good, while those printed right to left have lots of stringing.

If we were able to manually set the printing order, this problem would be solved easily.

I know there are other ways to solve the problem, e.g. enabling "external perimeters first" or "wipe while retracting", but sometimes those are not options for other reasons (I use firmware retraction - automatic detection, which doesn't work well with wiping).

rdrcrmatt, 2026-09-02:
This enhancement would be very useful in PrusaSlicer 3, please do not close.

closed 2026-10-06 by Jan-Soustruznik