prusa3d/PrusaSlicer #7319: Supports interleaved with shape body when slicing adjacent, separate bodies

drott, 2021-11-20:
### Version

2.4.0-beta1-arm64

### Operating system type + version

M1 Mac OS 12.0.1 (21A559)

### 3D printer brand / version + firmware version (if known)

Prusa Original i3 MK3S+

### Behavior

Load project file, slice with configured settings. 

Observe green supports material interleaving and crosstalking into the body at the section of the part where this rounded screw support channel is. 

Expected: Supports not crosstalking into part, and actually no supports are needed here.

Actual: Support structures crosstalking into part.

[image]

[image]

Other observations: Small seams are visible between the screw channel and the rest of the part in the 3D isometric view - this may be a fusion 3MF export issue? In a second attempt I joined the screw channel with the rest of the body vs. leaving it as a separate body in the original model, which avoids this slicing bug. Attaching the fusion project as well. 

#### Project File (.3MF) where problem occurs

[supports_in_body.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/7574447/supports_in_body.3mf.zip)

[Light Ceiling Cap v3.f3d.zip](https://github.com/prusa3d/PrusaSlicer/files/7574448/Light.Ceiling.Cap.v3.f3d.zip)

Area5142, 2021-11-20:
In the project _supports_in_body.3mf.zip_ there are two overlapping bodies. PrusaSlicer will treat them as individual bodies and generate g-code for each. The resulting g-code is overlapping (unwanted).

Select both bodies and select merge, now the resulting merged body will slice as expected.

lukasmatena, 2021-11-20:
Duplicate of #4517, #2501, #4788. As of now, there is no collision detection of individual objects, it is up to the user to prevent them. Merging the object so they become parts of a single object solves it (as mentioned by @Area5142). Closing.

closed 2021-11-20 by lukasmatena

drott, 2021-11-22:
Thanks for the quick triage. It seems like there is a difference in the STL export and the 3MF export of Fusion 360 in that the STL export would usually combine objects and create one joint mesh that doesn't expose this problem, while the 3MF export keeps bodies separate and thus triggers this slicing problem. Good to know and good to keep an eye out for and make sure to join bodies where needed before exporting.